f4exb/sdrangel #1297: 3D waterfall: wrap around issue

f4exb, 2022-06-21:
It seems there is a wrap around issue in the display. The oldest line of data is copied over the first (newest) line. This screenshot illustrates this:
[image]
Note: this is a single burst so the slice at the top is a ghost.

srcejon, 2022-06-21:
The problem, I think, is that when textures are set to wrap, OpenGL interpolates the value at texture coordinate near to 1 with the value at texture coordinate 0. i.e. old data is getting interpolated with new data.

In m_vertexShader, I added the clamp function:

    "   coord = textureTransform * vec4(clamp(coord2d, limit, 1.0-limit), 0, 1);\n"

Where limit = 1.5f*1.0f/(float)(m_gridElements)   (and gridElements is fftSize)

To try to prevent the texture coordinate from being close enough to 0 or 1, where this interpolation might occur. That fixed the problem on my graphics card (RTX 3080) as far as I could see.

The code for the 2D waterfall does a similar sort of trick to limit the max texture coordinate, with:

           float off = 1.0 / (m_waterfallTextureHeight - 1.0);

If you set off=0, and disable the drawing of the rect around the 2D waterfall (so you can see the first line of pixels), you can see the same problem occur on the 2D waterfall.

It could be a difference between the filtering in different graphics cards, but perhaps it just depends on the waterfall height. The clamp is using the fft size, rather than m_waterfallTextureHeight - so perhaps I need to change that.

Another thing that might be an idea is using QOpenGLTexture::Nearest rather than QOpenGLTexture::Linear - but this would affect pixels in the middle of the texture as well.

f4exb, 2022-06-21:
Works with NVIDIA GTX1070 as well.

closed 2022-06-21 by f4exb