f4exb/sdrangel #1308: Option to disable compiling named plug-ins

joecupano, 2022-06-24:
Per subject

f4exb, 2022-06-24:
That's a bit slim... can you elaborate?

srcejon, 2022-06-24:
I'm guessing -DENABLE_APT_DEMOD=OFF etc.

joecupano, 2022-06-25:
Good example. Disable specific plugins

f4exb, 2022-06-25:
There are currently 61 plugins in channelrx, channeltx and features. For the "sampling" plugins there are already enablers.

closed 2022-06-25 by f4exb

f4exb, 2022-06-25:
I would advise using the cmake user profiles rather than having extra long -D series. This is an example of an "ultra minimalistic" profile supporting only RTL SDR and a few audio Rx channel plugins:

```json
{
    "version": 3,
    "configurePresets": [
        {
            "name": "ultraminimalist",
            "inherits": "default",
            "cacheVariables": {
                "BUILD_SERVER": "OFF",
                "ENABLE_AIRSPY":  "OFF",
                "ENABLE_AIRSPYHF":  "OFF",
                "ENABLE_BLADERF":  "OFF",
                "ENABLE_FUNCUBE":  "OFF",
                "ENABLE_HACKRF":  "OFF",
                "ENABLE_IIO":  "OFF",
                "ENABLE_LIMESUITE":  "OFF",
                "ENABLE_MIRISDR":  "OFF",
                "ENABLE_PERSEUS":  "OFF",
                "ENABLE_RTLSDR": "ON",
                "ENABLE_SDRPLAY":  "OFF",
                "ENABLE_SOAPYSDR":  "OFF",
                "ENABLE_USRP":  "OFF",
                "ENABLE_XTRX":  "OFF",
                "WITH_DOC": "OFF",
                "ENABLE_CHANNELMIMO": "OFF",
                "ENABLE_CHANNELTX": "OFF",
                "ENABLE_FEATURE": "OFF",
                "ENABLE_CHANNELRX": "ON",
                "ENABLE_CHANNELRX_DEMODADSB": "OFF",
                "ENABLE_CHANNELRX_DEMODAM": "ON",
                "ENABLE_CHANNELRX_DEMODDATV": "OFF",
                "ENABLE_CHANNELRX_RADIOCLOCK": "OFF",
                "ENABLE_CHANNELRX_RADIOASTRONOMY": "OFF",
                "ENABLE_CHANNELRX_DEMODRADIOSONDE": "OFF",
                "ENABLE_CHANNELRX_FREQTRACKER": "OFF",
                "ENABLE_CHANNELRX_DEMODATV": "OFF",
                "ENABLE_CHANNELRX_DEMODVORMC": "OFF",
                "ENABLE_CHANNELRX_DEMODPAGER": "OFF",
                "ENABLE_CHANNELRX_DEMODDAB": "OFF",
                "ENABLE_CHANNELRX_UDPSINK": "OFF",
                "ENABLE_CHANNELRX_DEMODAIS": "OFF",
                "ENABLE_CHANNELRX_DEMODNFM": "ON",
                "ENABLE_CHANNELRX_FILESINK": "OFF",
                "ENABLE_CHANNELRX_DEMODFREEDV": "OFF",
                "ENABLE_CHANNELRX_DEMODCHIRPCHAT": "OFF",
                "ENABLE_CHANNELRX_REMOTESINK": "OFF",
                "ENABLE_CHANNELRX_DEMODSSB": "ON",
                "ENABLE_CHANNELRX_CHANALYZER": "OFF",
                "ENABLE_CHANNELRX_SIGMFFILESINK": "OFF",
                "ENABLE_CHANNELRX_DEMODBFM": "ON",
                "ENABLE_CHANNELRX_DEMODWFM": "ON",
                "ENABLE_CHANNELRX_NOISEFIGURE": "OFF",
                "ENABLE_CHANNELRX_DEMODVOR": "OFF",
                "ENABLE_CHANNELRX_LOCALSINK": "OFF",
                "ENABLE_CHANNELRX_DEMODPACKET": "OFF",
                "ENABLE_CHANNELRX_DEMODAPT": "OFF",
                "ENABLE_CHANNELRX_DEMODDSD": "OFF"
            },
            "binaryDir": "build-ultraminimalist"
        }
    ],
    "buildPresets": [
        {
            "name": "my-default",
            "inherits": "default",
            "jobs": 12
        },
        {
            "name": "ultraminimalist",
            "configurePreset": "ultraminimalist",
            "jobs": 12
        }
    ]
}
```
It builds up on the ¨default" profile that has been added that just specifies the dependencies locations from the WiKi you may just start your own instead.

Note that you have to specify `"ENABLE_CHANNELRX": "ON"` although it is the default.

This simple Python script will make the list with "OFF" values when you start it from the plugins section directory (e.g. sdrangel/plugins/channelrx):

```python
#! /usr/bin/env python
import os
import fnmatch

for path,dirs,files in os.walk('.'):
    for f in fnmatch.filter(files,'*plugin.cpp'):
        plugin_name = os.path.basename(path)
        plugins_dir = os.path.split(os.path.split(os.path.abspath(path))[0])[1]
        print(f'"ENABLE_{plugins_dir.upper()}_{plugin_name.upper()}": "OFF",')

```

f4exb, 2022-06-25:
On my machine with `make -j12` this "ultra minimalist" preset compiles in ~6 minutes and a full compile in ~11 minutes. There can be a significant gain in compile time if your intention is to build a version targeting very specific needs needing only a few plugins. There is of course some gain at program start time as well.

f4exb, 2022-06-25:
By fixing #1310 `"ENABLE_CHANNELRX": "ON"` should not be needed anymore

commit d2951627 (2022-06-25): Added cmake enablers for all non sampling device plugins. Fixes #1308