f4exb/sdrangel #1310: No available features on Ubuntu 22.04

Baa14453, 2022-06-25:
I compiled SDRAngel on Ubuntu 22.04 and when I try to add a feature I cannot open the drop-down, unsure if the drop-down menu is not working or if there are no features available.

[image]

Other drop-down menus work fine:

[image]

I followed the compile guide []here(https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) with no hardware dependencies added.

I'm using Ubuntu 22.04 with Wayland.

Baa14453, 2022-06-25:
I've compiled this twice, once on my PC and once within a virtual machine, both re-created the problem, however with my own PC some of the features have now appeared, although I don't think I changed anything besides launching SDRAngel again from a new terminal.

Some features are still missing though, such as the Pager Demodulator, this might be unrelated:

[image]

In my virtual machine the issue persists so I can still test suggestions.

srcejon, 2022-06-25:
The Pager Demod is a channel, not a Feature. So you add that from the Device window.

Baa14453, 2022-06-25:
> The Pager Demod is a channel, not a Feature. So you add that from the Device window.

Thanks I looked everywhere for the Channels button but couldn't see it anywhere, found it just next to the device name finally:
[image]

Baa14453, 2022-06-25:
Similar to before in my virtual machine where Features are not appearing, the Channels list is also empty 

[image]

srcejon, 2022-06-25:
When did you get the latest source code?

Could be related to this change: https://github.com/f4exb/sdrangel/issues/1308

Your log only has device plugins loaded - no features / channels.

Baa14453, 2022-06-25:
> When did you get the latest source code?

I cloned a couple hours ago i'll try the commit before

Baa14453, 2022-06-25:
> When did you get the latest source code?
> 
> Could be related to this change: #1308
> 
> Your log only has device plugins loaded - no features / channels.

Just compiled commit `58328ace0d04f662dea1cb2c719573902f7e77a6` and it worked no problem, it also resolved #1311 I'll try it outside my virtual machine now and see if it still works.

f4exb, 2022-06-25:
There is a stray comma here: https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L51
...and here: https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L82
...and here: https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L82
...and here: https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L106

This is probably the issue.

commit 980c7a98 (2022-06-25): Removed stray comma in CMakeLists.txt. Fixes #1310
closed 2022-06-25 by f4exb