f4exb/sdrangel #1311: Adding any channel causes Segmentation Fault in Ubuntu 22.04

Baa14453, 2022-06-25:
I've tried this with the Hack RF rx and the Audio rx, if I add any channel SDRAngel will crash with `Segmentation fault (core dumped)`.
I compiled SDRAngel earlier today.

Running Ubuntu 22.04 with Wayland.

In this example I used the Audio Input with the ABS channel:
```
2022-06-25 16:11:17.535 (I) Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
2022-06-25 16:11:17.667 (I) SDRangel 7.3.2-60-g77963f9eb Qt 5.15.3 64b x86_64 Ubuntu 22.04 LTS DSP Rx:24b Tx:16b PID 65276
2022-06-25 16:11:17.667 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/baa/.config/f4exb/SDRangel.conf
2022-06-25 16:11:17.668 (D) MessagePipes::startGC
2022-06-25 16:11:17.668 (D) DataPipes::startGC
2022-06-25 16:11:17.668 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #1: samplerate
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #2: speexrate
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #3: jack
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #4: oss
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #5: pulse
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #6: upmix
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #7: vdownmix
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #8: usbstream:CARD=HDMI
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #9: hw:CARD=CODEC,DEV=0
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #10: plughw:CARD=CODEC,DEV=0
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #11: sysdefault:CARD=CODEC
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #12: front:CARD=CODEC,DEV=0
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #13: dsnoop:CARD=CODEC,DEV=0
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #14: usbstream:CARD=CODEC
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #15: alsa_input.usb-Burr-Brown_from_TI_USB_Audio_CODEC-00.analog-stereo-input
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #16: alsa_output.usb-Burr-Brown_from_TI_USB_Audio_CODEC-00.analog-stereo-output.monitor
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: input device #17: alsa_output.pci-0000_2f_00.1.hdmi-stereo-extra2.monitor
2022-06-25 16:11:17.807 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #0: default
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #1: samplerate
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #2: speexrate
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #3: jack
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #4: oss
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #5: pulse
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #6: upmix
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #7: vdownmix
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #8: hw:CARD=HDMI,DEV=3
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #9: hw:CARD=HDMI,DEV=7
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #10: hw:CARD=HDMI,DEV=8
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #11: hw:CARD=HDMI,DEV=9
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #12: hw:CARD=HDMI,DEV=10
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #13: hw:CARD=HDMI,DEV=11
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #14: plughw:CARD=HDMI,DEV=3
2022-06-25 16:11:18.232 (D) AudioDeviceManager::AudioDeviceManager: output device #15: plughw:CARD=HDMI,DEV=7
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #16: plughw:CARD=HDMI,DEV=8
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #17: plughw:CARD=HDMI,DEV=9
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #18: plughw:CARD=HDMI,DEV=10
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #19: plughw:CARD=HDMI,DEV=11
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #20: hdmi:CARD=HDMI,DEV=0
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #21: hdmi:CARD=HDMI,DEV=1
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #22: hdmi:CARD=HDMI,DEV=2
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #23: hdmi:CARD=HDMI,DEV=3
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #24: hdmi:CARD=HDMI,DEV=4
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #25: hdmi:CARD=HDMI,DEV=5
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #26: dmix:CARD=HDMI,DEV=3
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #27: dmix:CARD=HDMI,DEV=7
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #28: dmix:CARD=HDMI,DEV=8
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #29: dmix:CARD=HDMI,DEV=9
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #30: dmix:CARD=HDMI,DEV=10
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #31: dmix:CARD=HDMI,DEV=11
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #32: usbstream:CARD=HDMI
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #33: hw:CARD=CODEC,DEV=0
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #34: plughw:CARD=CODEC,DEV=0
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #35: sysdefault:CARD=CODEC
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #36: front:CARD=CODEC,DEV=0
2022-06-25 16:11:18.233 (D) AudioDeviceManager::AudioDeviceManager: output device #37: surround21:CARD=CODEC,DEV=0
[... 340 lines cut ...]
2022-06-25 16:11:22.285 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2022-06-25 16:11:22.285 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2022-06-25 16:11:22.285 (D) GLShaderSpectrogram::initializeGL: m_useImmutableStorage:  true
2022-06-25 16:11:22.286 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2022-06-25 16:11:22.286 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2022-06-25 16:11:22.291 (D) GLShaderSpectrogram::initGrid: requested:  512  actual:  512
2022-06-25 16:11:22.306 (D) GLShaderSpectrogram::initGrid: requested:  512  actual:  512
2022-06-25 16:11:22.311 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  0  sampleRate:  0
2022-06-25 16:11:22.311 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  12000  sampleRate:  48000
2022-06-25 16:11:22.311 (D) AudioInputGui::handleInputMessages: message: DSPSignalNotification
2022-06-25 16:11:22.311 (D) AudioInputGui::handleInputMessages: DSPSignalNotification: SampleRate: 48000
2022-06-25 16:11:22.311 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2022-06-25 16:11:24.464 (D) ADSBDemod::ADSBDemod
2022-06-25 16:11:24.464 (D) ADSBDemodSink::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_correlationThreshold:  10  m_correlateFullPreamble:  true  m_demodModeS:  false  m_samplesPerBit:  4  force:  true
2022-06-25 16:11:24.464 (D) Interpolator::createPolyphaseLowPass: ntaps: 14
[... cut ...]
2022-06-25 16:11:24.473 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2022-06-25 16:11:24.473 (D) DSPDeviceSourceEngine::addSink:  ADSBDemod
2022-06-25 16:11:24.473 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2022-06-25 16:11:24.473 (D) ChannelGUI::ChannelGUI
Segmentation fault (core dumped)
```

Baa14453, 2022-06-25:
This is probably fixed by https://github.com/f4exb/sdrangel/commit/980c7a98f21b26a3fbcf25f0d5a292cfa148ecd0 but worth noting that disabling ChannelRX Plugins and then trying to add one might cause Sdrangel to crash, i'll test it tomorrow

f4exb, 2022-06-26:
There used to be this issue when trying to add channel from an empty list but this has been fixed by 77963f9ebf6493b36d2c782dce495ad61c6f647b before the code for #1308 was pushed to master.

closed 2022-08-30 by github-actions[bot]