f4exb/sdrangel #1312: Best SW for SDRs but some defects.

mistergitj, 2022-06-27:
I am running V7.4.0 under W10 with a AirSpy for the last couple of months. I have several years of experience with other SW and other SDRs.  SDRangel is by far the best I have encountered but it does have some problems.  All my experience is on V7.  Lately it crashes way too often, it almost always crashes when closing.  This is a problem when I have updated a Device Set Preset-my latest updates are lost. Clicking the X on the big window or the Workspace window gives a rotating blue circle and then the big window closes after a few seconds.
I use the ? often but it often fails.  AirSpy main window does nothing when ? is clicked. Spectrum Marker dialog and several others yield a red circle with a slash through it.
Please tell me what saved information may be useful for debug.
Thanks much.  Enjoy, John.

srcejon, 2022-06-28:
>  Lately it crashes way too often, it almost always crashes when closing. 

What features / channels do you have open when it crashes? A screen shot might help

>This is a problem when I have updated a Device Set Preset-my latest updates are lost.

Preferences > Configurations > Update button should save the config before you exit.

> AirSpy main window does nothing when ? is clicked.

The link for this is missing. I'll add it.

>Spectrum Marker dialog and several others yield a red circle with a slash through it. 

Can you post a screenshot showing this please?

choupette-change, 2022-06-28:
I also have problems with blocks and slowness especially when displaying and using 3D maps (especially with the Ubuntu version). My Internet connection is OK. But I think the power of the PC must be high to be able to use SDRangel in the latest versions (v7.4.0). What type of PC configuration do you recommend (RAM, number of cores, clock frequency ...) ? My current PC must be too old... (Google traduction)

f4exb, 2022-06-28:
> Spectrum Marker dialog and several others yield a red circle with a slash through it.

I cannot take a screenshot of the "no way" sign that appears but I think it is when the "?" is clicked in the top bar of the system dialogs like:
[image]

I don't know how a link to a web page can be inserted there...

srcejon, 2022-06-28:
In HttpDownloadManagerGUI I get rid of it with:

        Qt::WindowFlags flags = progressDialog->windowFlags();
        flags &= ~Qt::WindowContextHelpButtonHint;
        progressDialog->setWindowFlags(flags);

If you want to handle it, stackoverflow suggests https://doc.qt.io/qt-5/qwidget.html#nativeEvent

f4exb, 2022-06-28:
Or:
```c++
setWindowFlags(windowFlags().setFlag(Qt::WindowContextHelpButtonHint, false))
```
Handling the event seems a bit complex for things documented elsewhere.

srcejon, 2022-06-28:
You might need Qt::CustomizeWindowHint as well.

mistergitj, 2022-06-28:
I will investigate and try all the suggested solutions soon. It does not seem to matter what Channels/Features are being used. I will post a screenshot as best I can. I also found I could not get a SS of the red circle/slash but will post a picture.

choupette-change, when I bragged about my machine having 16 cores capable of running 32 threads, I was told that the current V7 does not multi-thread.  Right now I am running ADS-B and seeing 23 threads running:
[image]
 Thanks much for all the quick response.  Enjoy, John.

srcejon, 2022-06-28:
> I was told that the current V7 does not multi-thread. 

V7 is multi-threaded - I was referring only to one function within the Radio Astronomy plugin that you was having problems with.

f4exb, 2022-07-01:
> Lately it crashes way too often, it almost always crashes when closing.

It does for me on Windows but not on Linux (Ubuntu 22.04). On Windows the segfault appears right after `DeviceGUI` is destroyed:

```
2022-06-30 19:24:04.569 (D) MainWindow::closeEvent
2022-06-30 19:24:04.570 (D) MainWindow::saveConfiguration: configuration [default | no name] 1 workspaces
2022-06-30 19:24:04.570 (D) DeviceUISet::saveDeviceSetSettings: preset: [default, no name]
2022-06-30 19:24:04.571 (D) DeviceUISet::saveRxChannelSettings: saving channel [sdrangel.channel.nfmdemod]
2022-06-30 19:24:04.571 (D) DeviceUISet::saveRxChannelSettings: saving channel [sdrangel.channel.nfmdemod]
2022-06-30 19:24:04.571 (D) DeviceAPI::saveSamplingDeviceSettings: serializing source sdrangel.samplesource.rtlsdr[0]: R8202005
2022-06-30 19:24:04.571 (D) MainWindow::saveConfiguration: sdrangel.samplesource.rtlsdr device in workspace 0 spectrum in 0
2022-06-30 19:24:04.572 (D) AudioDeviceManager::serialize
2022-06-30 19:24:04.573 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "System default device"  sampleRate:  48000  volume:  0.15
2022-06-30 19:24:04.573 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  true  udpUseRTP:  false  udpChannelMode:  2  udpChannelCodec:  5  decimationFactor:  1
2022-06-30 19:24:04.594 (D) DSPDeviceSourceEngine::stopAcquistion
2022-06-30 19:24:04.595 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-06-30 19:24:04.595 (D) DSPDeviceSourceEngine::gotoIdle
2022-06-30 19:24:04.596 (C) RTLSDRThread: async error: No error
2022-06-30 19:24:04.597 (D) GLSpectrumGUI::handleInputMessages: message: SpectrumVis::MsgStartStop
2022-06-30 19:24:04.599 (D) NFMDemod::stop
2022-06-30 19:24:04.599 (D) NFMDemod::stop
2022-06-30 19:24:04.601 (D) DC offset:0.000000,0.000000
2022-06-30 19:24:04.601 (D) DSPDeviceSourceEngine::removeSink:  SpectrumVis
2022-06-30 19:24:04.601 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2022-06-30 19:24:04.601 (D) DeviceUISet::freeChannels: destroying channel [sdrangel.channel.nfmdemod]
2022-06-30 19:24:04.601 (D) DSPDeviceSourceEngine::removeSink:  NFMDemod
2022-06-30 19:24:04.602 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2022-06-30 19:24:04.602 (D) AudioDeviceManager::removeAudioSink: 0x222ada71200
2022-06-30 19:24:04.603 (D) ChannelGUI::~ChannelGUI
2022-06-30 19:24:04.607 (D) ChannelGUI::~ChannelGUI: end
2022-06-30 19:24:04.623 (D) DeviceUISet::freeChannels: destroying channel [sdrangel.channel.nfmdemod]
2022-06-30 19:24:04.623 (D) DSPDeviceSourceEngine::removeSink:  NFMDemod
2022-06-30 19:24:04.624 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2022-06-30 19:24:04.625 (D) AudioDeviceManager::removeAudioSink: 0x222ae449e30
2022-06-30 19:24:04.627 (D) ChannelGUI::~ChannelGUI
2022-06-30 19:24:04.632 (D) ChannelGUI::~ChannelGUI: end
2022-06-30 19:24:04.651 (D) RTLSDRGui::~RTLSDRGui
2022-06-30 19:24:04.651 (D) RTLSDRGui::~RTLSDRGui: end
2022-06-30 19:24:04.651 (D) DeviceGUI::~DeviceGUI
2022-06-30 19:24:04.655 (D) DeviceGUI::~DeviceGUI: end
Segmentation fault
```

In comparison when closing the main window on Linux the process goes past this and enters the `MainSpectrumGUI` destructor:
```
...
2022-06-30 21:37:10.814 (D) ChannelGUI::~ChannelGUI
2022-06-30 21:37:10.814 (D) ChannelGUI::~ChannelGUI: end
2022-06-30 21:37:10.815 (D) RTLSDRGui::~RTLSDRGui
2022-06-30 21:37:10.815 (D) RTLSDRGui::~RTLSDRGui: end
2022-06-30 21:37:10.815 (D) DeviceGUI::~DeviceGUI
2022-06-30 21:37:10.815 (D) DeviceGUI::~DeviceGUI: end
Reattached kernel driver
2022-06-30 21:37:10.962 (D) MainSpectrumGUI::~MainSpectrumGUI
2022-06-30 21:37:10.962 (D) MainSpectrumGUI::~MainSpectrumGUI: end
2022-06-30 21:37:11.096 (D) DSPDeviceSourceEngine::stop
...
```

So I am guessing something specific to Windows is happening in `~MainSpectrumGUI`

mistergitj, 2022-07-01:
Thanks much, f4exb.  I was afraid of that.  I will look in my system log and post what looks relevant.  Enjoy, John.

mistergitj, 2022-07-01:
Here is an example, will look some more later, but a quick look were copies of this one:

- System 

  - Provider 

   [ Name]  Application Error 
 
  - EventID 1000 

   [ Qualifiers]  0 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x80000000000000 
 
  - TimeCreated 

   [ SystemTime]  2022-06-30T20:56:01.4433924Z 
 
   EventRecordID 29074 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  0 
   [ ThreadID]  0 
 
   Channel Application 
 
   Computer ?
 
   Security 
 

- EventData 

   sdrangel.exe 
   0.0.0.0 
   62b7f2b6 
   libusb-1.0.dll 
   1.0.26.11724 
   6252cb06 
   c0000005 
   0000000000001d05 
   22f4 
   01d88c8257edfcc2 
   C:\Program Files\SDRangel\sdrangel.exe 
   C:\Program Files\SDRangel\libusb-1.0.dll 
   2fba1fff-9b9a-4435-9177-fe696983cf85 
    

Thanks and enjoy, John.

mistergitj, 2022-07-03:
I have discovered that if I do not Start the SDR, then I can close SDRangel with no crash. But if I click Start button, then SDRangel crashes every time on close. The Channels/Features running have no effect. Thanks and enjoy, John.

mistergitj, 2022-07-19:
Initial testing of 7.5.1 indicates it has the same problems for me as previous versions. Thanks, John.

closed 2022-09-23 by github-actions[bot]