f4exb/sdrangel #1321: double free or corruption (!prev)

eu2aa, 2022-07-01:
Hello Edouard,

Thanks for the great program.
I have been using it every day for over 3 years.
Thanks to your program, I achieved high results when working through the QO-100 satellite in the CW mode
https://qo100dx.club/rankings-dxcc
https://qo100dx.club/rankings-grids
using an earlier version.

I also work in SSB and FT-8 (I keep PTT manually).

A week ago I switched to your latest version of SDRangel 7.3.0, and see many new features.

Unfortunately, getting frequent crashes. Literally every 2-4 minutes or so.
In SSB, CW and FT-8 modes.
Sometimes I can finish QSO, and sometimes I can't.

My setup: OpenSUSE + PLUTO + SDRangel 7.3.0

I am getting this error:

```
double free or corruption (!prev)
sdrangel exited with code: null
```

I would be very grateful for any advice,

--
Yours sincerely,
Vladimir Chepyzhenko,
EU2AA
2019 - https://observablehq.com/@eu2aa
2017  - http://eu2aa.com
1992 - https://eu2aa-qrz-ru.translate.goog/orbita.html?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_sch=http

srcejon, 2022-07-02:
Which was the earlier version that was working for you?

A screenshot showing the channels / features you have open might help.

eu2aa, 2022-07-02:
I don't remember exact version, but it was v6.x.x

[image]

Sometimes it gives this splash screen.

[image]

eu2aa, 2022-07-05:
free(): invalid pointer
node:internal/errors:692
  const ex = new Error(msg);
             ^

Error: socket hang up
    at connResetException (node:internal/errors:692:14)
    at Socket.socketOnEnd (node:_http_client:478:23)
    at Socket.emit (node:events:539:35)
    at endReadableNT (node:internal/streams/readable:1345:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  code: 'ECONNRESET'
}

eu2aa, 2022-07-17:
double free or corruption (!prev)
sdrangel exited with code: null

eu2aa, 2022-07-17:
Took action:
1 Connected PLUTO through a separate USB-C HUB.
2 MIDI, which I use for the RIT function, remained on the same HUB.
A USB microphone is also connected there.
3 I noticed that when TX1 was turned on, the SDRangel screen began to twitch much less.

eu2aa, 2022-08-05:
Noticed one feature.
First of all, the panorama freezes, the reception continues, but blindly.
There is no transmission either.

eu2aa, 2022-08-05:
On August 1, the CW worked with the F6HCC.
The program crashed 6 times.
Jean was perplexed, had to apologize every time.

eu2aa, 2022-09-27:
**Solving the problem**

After starting Strangel in the window
https://github.com/eu2aa/eu2aa.github.io/blob/master/photo/T1_.png 
you need to press the start/stop generation button THREE TIMES: ON — OFF — ON

Then in the window
https://github.com/eu2aa/eu2aa.github.io/blob/master/photo/T1.0.png 
you need to press the Audio input button TWICE: OFF — ON

There are no failures after these procedures.

**The problem is solved.**

closed 2022-12-03 by github-actions[bot]