f4exb/sdrangel #1324: PlutoSDR RX Max Freq. 5749987

eu2aa, 2022-07-02:
While working with PlutoSDR I was able to tune TX up to 6GHz while RX is limited to 5.749.987.000
Is it limitation of SDRangel or PlutoSDR?

[image]

drom, 2022-07-02:
I have the same issue. PlutoSDR customized as ad9361 https://wiki.analog.com/university/tools/pluto/users/customizing
Suppose to support up to 6GHz. But somehow is limited to 5,749,987 in SDRangel.

f4exb, 2022-07-03:
Look at that "X" button that is engaged. This is the transverter mode. This recalculates the dial frequency depending on the transverter shift value. You have to switch it off for a direct access to the device frequency.

drom, 2022-07-03:
Yep. Thank you. That works

eu2aa, 2022-07-03:
В письме от воскресенье, 3 июля 2022 г. 10:40:58 MSK 
пользователь Edouard Griffiths написал:
> Look at that "X" button that is engaged. This is the 
transverter mode. This recalculates the dial frequency 
depending on the transverter shift value. You have to switch it 
off for a direct access to the device frequency.

Yes, I disabled the X button, the frequency was adjusted higher! 
Thanks for the helpful advice Edouard!

f4exb, 2022-07-03:
hence closed.

closed 2022-07-03 by f4exb