f4exb/sdrangel #1327: Airspy in Macos

loewal, 2022-07-07:
Airspy HF+ and Discovery don't work in Macos, neither selfcompiled nor DMG.
Windows and Linux work fine.

loewal, 2022-07-18:
May I express my disappointment for not having recieved any reaction on my problem?

The Airspy's are found, but do NOTHING

lawendel, 2022-08-19:
I'm observing the same (lack of) behavior on a MacBook Air M1 under Monterey. I've downloaded the i64 DMG and installed flawlessly. The program opens with no error messages, it then recognizes the HF+ as an input device, it can even switch between HF and VHF, but nothing happens in the Spectrum window. Before reading your issue, I thought I made something wrong. Maybe it's that... Is there a mailing list or a group where we can ask directly to f4exb? SDRangel is indeed complex but highly attractive as a cross-platform app. I'm also using GQRX and SDR++ with no such issues.

lawendel, 2022-08-19:
Sorry, it's the recent "macx64" DMG...

justfalter, 2022-08-28:
I have an Airspy r2 on an M1, compiled for arm64. 

A single breakpoint set in [AirspyWorker::rx_callback](https://github.com/f4exb/sdrangel/blob/d1a4fca49bcabf847f36b595f6b47202895d8080/plugins/samplesource/airspy/airspyworker.cpp#L279-L291) would never hit after starting the input.

After some debugging,  I found I could get the Airspy input working properly by:
1. Setting a breakpoint right before the `m_airspyWorkerThread` is started [here](https://github.com/f4exb/sdrangel/blob/d1a4fca49bcabf847f36b595f6b47202895d8080/plugins/samplesource/airspy/airspyinput.cpp#L240)
2. Starting the input
3. Hit the breakpoint
4. Wait a couple seconds
5. Continue execution

My guess is that there is some kind of timing issue in the worker thread startup sequence.

The effective sequence is:
* `m_airspyWorker->startWork()`
  * `airspy_stop_rx(...)`
  *  (on success): `airspy_is_streaming(...)`
* `m_airspyWorkerThread.start()`

I'm unfamiliar with the mechanics of a QThread, as that is how [m_airspyWorkerThread is defined](https://github.com/f4exb/sdrangel/blob/d1a4fca49bcabf847f36b595f6b47202895d8080/plugins/samplesource/airspy/airspyinput.h#L142).

https://github.com/f4exb/sdrangel/blob/d1a4fca49bcabf847f36b595f6b47202895d8080/plugins/samplesource/airspy/airspyinput.cpp#L236-L247

https://github.com/f4exb/sdrangel/blob/d1a4fca49bcabf847f36b595f6b47202895d8080/plugins/samplesource/airspy/airspyworker.cpp#L45-L62

justfalter, 2022-08-28:
Note

> After some debugging, I found I could get the Airspy input working properly by:
>
> 1. Setting a breakpoint right before the `m_airspyWorkerThread` is started [...]

Note: this seems to only work with a debugger breakpoint, so far. I've tried sleeping just before the `m_airspyWorkerThread.start()` call and it does not result in the airspy working. I tried with both `               QThread::msleep(2000);` and `std::this_thread::sleep_for(std::chrono::seconds(2))`, and neither resulted in the airspy r2 working.

I'm thinking that the issue might exist in a separate thread, as pausing on a breakpoint results in all threads being paused.

srcejon, 2022-09-20:
I just tried the airspyhf_rx example program that is included with airspyhf library, and that doesn't work for me on MacOS (Big Sur). Does it work for anyone else?

loewal, 2022-09-21:
> Do you have a USB 2 hub to plug it in to? I remember a previous issue where it wouldn't work with Linux with USB 3 ports, but would with USB 2.

I have.

Tried:

• First session with BWFM: nothing
• Second session: cannot start sample source

Why does it work with GQRX and SDR++……?

srcejon, 2022-09-21:
Well, SDR++ doesn't work for me either. Haven't tried GQRX yet.

srcejon, 2022-09-21:
Have opened an issue for airspyhf https://github.com/airspy/airspyhf/issues/42

loewal, 2022-09-21:
Airspy HF+ doesn’t work with the usb-2 hub, but works when plugged in directly in one of the Mac-ports.

loewal, 2022-09-21:
Sorry, it works on SDR++ with usb-2, when I give it some power ….duh….
Not on SDRangel.

closed 2022-09-27 by srcejon