f4exb/sdrangel #1329: webapiSettingsPutPatch multi-threading problem

srcejon, 2022-07-08:
I think there are a couple of multi-threading problems with webapiSettingsPutPatch in devices and channels.

Using AirspyHF code as an example, but the same applies elsewhere:

        int AirspyHFInput::webapiSettingsPutPatch(
                        bool force,
                        const QStringList& deviceSettingsKeys,
                        SWGSDRangel::SWGDeviceSettings& response, // query + response
                        QString& errorMessage)
        {
            (void) errorMessage;
            AirspyHFSettings settings = m_settings;
            webapiUpdateDeviceSettings(settings, deviceSettingsKeys, response);
        
            MsgConfigureAirspyHF *msg = MsgConfigureAirspyHF::create(settings, force);
            m_inputMessageQueue.push(msg);
        
            if (m_guiMessageQueue) // forward to GUI if any
            {
                MsgConfigureAirspyHF *msgToGUI = MsgConfigureAirspyHF::create(settings, force);
                m_guiMessageQueue->push(msgToGUI);
            }
        
            webapiFormatDeviceSettings(response, settings);
            return 200;
        }

Possible problems include:

     - No mutex is used. This can be problematic  for settings = m_settings, as this assignment isn't atomic - so if m_settings is being simultaneously written in applySettings(), settings may contain a mixture of old and new values.
     - If webapiSettingsPutPatch is called quickly twice in succession, handling of the Configure message pushed to the queue may not have been handled by the time of the second call - and thus settings will contain old settings. This can result in some settings being lost, if not all keys are specified in each call. (E.g. if there's one patch to log2Decim followed by another patch to devSampleRate, the setting for log2Decim can be lost).
     - Similarly, if webapiSettingsGet is called quickly after webapiSettingsPutPatch, old data may be returned for the get.

srcejon, 2022-07-08:
> For the first bullet this is very unlikely to happen and not worth a mutex.

There is already a mutex which I think should be suitable, just a case of adding:

    QMutexLocker mutexLocker(&m_mutex);

> For the middle bullet the solution is to return 204 that just says the messages have been processed. This is essentially asynchronous and we can't wait for the messages to be processed before returning the reply.

It's reasonable that processing is async, but how is a client to know when it is safe to send the next patch request?

> Also for the last bullet the process being essentially asynchronous this is something to cope with.

How though? Keep calling get until the patched value is returned?

f4exb, 2022-07-08:
This is a one way communication and how REST APIs work and one has to live with it.

srcejon, 2022-07-08:
A possible solution after a little experimenting:

- Add a m_semaphore member to the class
- Call m_semaphore.acquire() at start of webapiSettingsPutPatch()
- Add release parameter to the Configure message and set release=true for message in webapiSettingsPutPatch, but not other uses
- After m_settings = settings in applySettings, if release parameter is true, call m_semaphore.release()

Then for the get:

- Call m_semaphore.acquire() at start of webapiSettingsGet()
- Call m_semaphore.release() at the end of webapiSettingsGet()

This seems to ensure both functions can't start until the previous settings from putPatch have been applied. Thoughts?

srcejon, 2022-07-08:
Here's an example of where it's currently not working for me:

- Lets say m_settings starts with log2Decim=2 and devSampleRate=5M
- A call to webapiSettingsPutPatch() is made with a single key, say log2Decim=1
- webapiSettingsPutPatch takes a copy of m_settings as settings and set log2Decim=1
- These settings (log2Decim=1 and devSampleRate=5M) are pushed to the queue
- Another call to webapiSettingsPutPatch() is made with a single key devSampleRate=6M
- The thread handling the queue hasn't run / completed yet, so when webapiSettingsPutPatch takes a copy of m_settings as settings, we get the old value for log2Decim
- So the settings pushed to the queue are log2Decim=2 (the old value) and devSampleRate=6M
- The message handler runs
- First call to applySettings sets log2Decim =1 and devSampleRate=5M
- Second call to applySettings set log2Decim=2 and devSampleRate=6M
- So the patch to log2Decim is lost (overwritten with old value)

The semaphore ensures that applySettings has been called before m_settings is copied to settings, so we get:

-  m_settings starts with log2Decim=2 and devSampleRate=5M
- A call to webapiSettingsPutPatch() is made with a single key, say log2Decim=1
- webapiSettingsPutPatch takes a copy of m_settings as settings and updates log2Decim=1
- These settings (log2Decim=1 and devSampleRate=5M) are pushed to the queue
- Another call to webapiSettingsPutPatch() is made with devSampleRate=6M

This is the same as before, so far, but then:

- The thread handling the queue hasn't run / completed yet, so semaphore.acquire() blocks
- The message handler runs
- First call to applySettings sets log2Decim =1 and devSampleRate=5M
- semaphore is released
- webapiSettingsPutPatch takes a copy of the new m_settings and updates devSampleRate to 6M
- The settings pushed to the queue are now log2Decim=1 (the new value) and devSampleRate=6M
- Second call to applySettings sets log2Decim=1 and devSampleRate=6M
- So we get the expected settings

f4exb, 2022-07-08:
So the issue is only with PATCH since with PUT all fields of the settings are updated anyway so the latest that comes in is right and that's the rule. With PATCH only some fields are supposed to be updated but the update message takes the whole settings as the reference making it a PUT without saying. And it takes the settings as they are at the time the request comes in which may be in the middle of the de-queuing of messages and thus possibly before the last queued request was processed. The GUI is also part of the game since it also has its own copy of the settings.

The semaphore solution is bad because it makes the message queue useless and blocks the requests processing. One should rather look at treating requests as transactions with only one valid copy of the settings similarly to database transactions.

This is a big redesign and I am not sure it is actually worth it since you have to hit the API really hard  or be in a massively parallel environment either multi process or multi user to be in such a situation. In most cases the previous message will have been process before the next request comes in. Have you already encountered this situation practically?

srcejon, 2022-07-09:
Yes. It's a real problem. I found it while debugging why some patch calls weren't working.

Note that depending on the SDR used, the problem can appear even without hitting the API paticulary hard, as applySettings can take a relatively long time (Eg SDRplayv3's applySettings can be slow as it waits for settings to be applied https://github.com/f4exb/sdrangel/commit/be2dbabafee76a199fdfe3b870f703d5c1795282  ).

f4exb, 2022-10-22:
<details>
<summary>Full diff for Airspy</summary>

```diff
diff --git a/plugins/samplesource/airspy/airspygui.cpp b/plugins/samplesource/airspy/airspygui.cpp
index d45c0b8f6..5e3e983b7 100644
--- a/plugins/samplesource/airspy/airspygui.cpp
+++ b/plugins/samplesource/airspy/airspygui.cpp
@@ -87,6 +87,7 @@ void AirspyGui::resetToDefaults()
 {
 	m_settings.resetToDefaults();
 	displaySettings();
+    m_forceSettings = true;
 	sendSettings();
 }
 
@@ -119,7 +120,13 @@ bool AirspyGui::handleMessage(const Message& message)
     if (AirspyInput::MsgConfigureAirspy::match(message))
     {
         const AirspyInput::MsgConfigureAirspy& cfg = (AirspyInput::MsgConfigureAirspy&) message;
-        m_settings = cfg.getSettings();
+
+        if (cfg.getForce()) {
+            m_settings = cfg.getSettings();
+        } else {
+            m_settings.applySettings(cfg.getSettingsKeys(), cfg.getSettings());
+        }
+
         blockApplySettings(true);
         displaySettings();
         blockApplySettings(false);
@@ -267,6 +274,7 @@ void AirspyGui::sendSettings()
 void AirspyGui::on_centerFrequency_changed(quint64 value)
 {
 	m_settings.m_centerFrequency = value * 1000;
+    m_settingsKeys.append("centerFrequency");
 	sendSettings();
 }
 
@@ -274,42 +282,49 @@ void AirspyGui::on_LOppm_valueChanged(int value)
 {
 	m_settings.m_LOppmTenths = value;
 	ui->LOppmText->setText(QString("%1").arg(QString::number(m_settings.m_LOppmTenths/10.0, 'f', 1)));
+    m_settingsKeys.append("LOppmTenths");
 	sendSettings();
 }
 
 void AirspyGui::on_dcOffset_toggled(bool checked)
 {
 	m_settings.m_dcBlock = checked;
+    m_settingsKeys.append("dcBlock");
 	sendSettings();
 }
 
 void AirspyGui::on_iqImbalance_toggled(bool checked)
 {
 	m_settings.m_iqCorrection = checked;
+    m_settingsKeys.append("iqCorrection");
 	sendSettings();
 }
 
 void AirspyGui::on_sampleRate_currentIndexChanged(int index)
 {
 	m_settings.m_devSampleRateIndex = index;
+    m_settingsKeys.append("devSampleRateIndex");
 	sendSettings();
 }
 
 void AirspyGui::on_biasT_stateChanged(int state)
 {
[... 610 lines cut ...]
+    QString getDebugString(const QStringList& settingsKeys, bool force=false) const;
 };
 
 #endif /* _AIRSPY_AIRSPYSETTINGS_H_ */
diff --git a/sdrgui/gui/scaleengine.cpp b/sdrgui/gui/scaleengine.cpp
index 673e5b8e0..68783e20e 100644
--- a/sdrgui/gui/scaleengine.cpp
+++ b/sdrgui/gui/scaleengine.cpp
@@ -30,12 +30,6 @@ static double trunc(double d)
 
 QString ScaleEngine::formatTick(double value, int decimalPlaces)
 {
-    if (m_truncated && (m_physicalUnit != Unit::Scientific))
-    {
-        // value = ((value * m_scale) - m_truncationValue) / m_scale;
-        qDebug("ScaleEngine::formatTick: value: %f decimalPlaces: %d m_scale: %f", value, decimalPlaces, m_scale);
-    }
-
 	if (m_physicalUnit != Unit::TimeHMS)
 	{
 	    if (m_physicalUnit == Unit::Scientific) {

```

</details>

f4exb, 2022-11-19:
<details>
<summary>Full diff for AFC feature</summary>

```diff
diff --git a/plugins/feature/afc/afc.cpp b/plugins/feature/afc/afc.cpp
index ba4c6097f..57bc499cf 100644
--- a/plugins/feature/afc/afc.cpp
+++ b/plugins/feature/afc/afc.cpp
@@ -118,7 +118,7 @@ void AFC::start()
     m_worker->setMessageQueueToGUI(getMessageQueueToGUI());
     m_thread->start();
 
-    AFCWorker::MsgConfigureAFCWorker *msg = AFCWorker::MsgConfigureAFCWorker::create(m_settings, true);
+    AFCWorker::MsgConfigureAFCWorker *msg = AFCWorker::MsgConfigureAFCWorker::create(m_settings, QList<QString>(), true);
     m_worker->getInputMessageQueue()->push(msg);
 
     m_state = StRunning;
@@ -146,7 +146,7 @@ bool AFC::handleMessage(const Message& cmd)
 	{
         MsgConfigureAFC& cfg = (MsgConfigureAFC&) cmd;
         qDebug() << "AFC::handleMessage: MsgConfigureAFC";
-        applySettings(cfg.getSettings(), cfg.getForce());
+        applySettings(cfg.getSettings(), cfg.getSettingsKeys(), cfg.getForce());
 
 		return true;
 	}
@@ -225,70 +225,32 @@ bool AFC::deserialize(const QByteArray& data)
 {
     if (m_settings.deserialize(data))
     {
-        MsgConfigureAFC *msg = MsgConfigureAFC::create(m_settings, true);
+        MsgConfigureAFC *msg = MsgConfigureAFC::create(m_settings, QList<QString>(), true);
         m_inputMessageQueue.push(msg);
         return true;
     }
     else
     {
         m_settings.resetToDefaults();
-        MsgConfigureAFC *msg = MsgConfigureAFC::create(m_settings, true);
+        MsgConfigureAFC *msg = MsgConfigureAFC::create(m_settings, QList<QString>(), true);
         m_inputMessageQueue.push(msg);
         return false;
     }
 }
 
-void AFC::applySettings(const AFCSettings& settings, bool force)
+void AFC::applySettings(const AFCSettings& settings, const QList<QString>& settingsKeys, bool force)
 {
-    qDebug() << "AFC::applySettings:"
-            << " m_title: " << settings.m_title
-            << " m_rgbColor: " << settings.m_rgbColor
-            << " m_trackerDeviceSetIndex: " << settings.m_trackerDeviceSetIndex
-            << " m_trackedDeviceSetIndex: " << settings.m_trackedDeviceSetIndex
-            << " m_hasTargetFrequency: " << settings.m_hasTargetFrequency
-            << " m_transverterTarget: " << settings.m_transverterTarget
-            << " m_targetFrequency: " << settings.m_targetFrequency
-            << " m_freqTolerance: " << settings.m_freqTolerance
-            << " m_trackerAdjustPeriod:" << settings.m_trackerAdjustPeriod
-            << " force: " << force;
+    qDebug() << "AFC::applySettings:" << settings.getDebugString(settingsKeys, force) << " force: " << force;
 
     QList<QString> reverseAPIKeys;
 
-    if ((m_settings.m_title != settings.m_title) || force) {
-        reverseAPIKeys.append("title");
-    }
-    if ((m_settings.m_rgbColor != settings.m_rgbColor) || force) {
-        reverseAPIKeys.append("rgbColor");
-    }
-    if ((m_settings.m_trackerDeviceSetIndex != settings.m_trackerDeviceSetIndex) || force) {
[... 445 lines cut ...]
 
     private:
         AFCSettings m_settings;
+        QList<QString> m_settingsKeys;
         bool m_force;
 
-        MsgConfigureAFCWorker(const AFCSettings& settings, bool force) :
+        MsgConfigureAFCWorker(const AFCSettings& settings, const QList<QString>& settingsKeys, bool force) :
             Message(),
             m_settings(settings),
+            m_settingsKeys(settingsKeys),
             m_force(force)
         { }
     };
@@ -154,7 +157,7 @@ private:
     QRecursiveMutex m_mutex;
 
     bool handleMessage(const Message& cmd);
-    void applySettings(const AFCSettings& settings, bool force = false);
+    void applySettings(const AFCSettings& settings, const QList<QString>& settingsKeys, bool force = false);
     void initTrackerDeviceSet(int deviceSetIndex);
     void initTrackedDeviceSet(int deviceSetIndex);
     void processChannelSettings(
```
</details>

closed 2026-02-01 by f4exb