f4exb/sdrangel #1330: Audio record

edombek, 2022-07-08:
is it possible to record audio?

srcejon, 2022-07-08:
Not directly within SDRangel, as far as I know. Have often thought about adding it to the Demod Analyzer feature.

You can pipe audio to a different program to record it - exact mechanism depends what OS you are on.

f4exb, 2022-07-09:
> Have often thought about adding it to the Demod Analyzer feature.

Or have a specific recorder feature so you do not have the overhead of the Demod Analyzer just to record audio.

However I think it is more appropriate in many situations to be able to record the audio mix (from all or a set of channels sharing the same audio device) and therefore resort to piping the audio output of the audio device you are interested in. It may be possible to tap audio at the mixing stage in SDRangel but in this case this has to be a global feature not a feature plugin. 

There is already the possibility to send the audio mix of an audio device (including virtual devices) via UDP/RTP so we may not be too far from being able to record this mix also:
[image]

Conversely if you want to record audio only from a specific channel plugin a recorder feature plugin may be more convenient than setting a dedicated virtual audio device. Also it allows to capture a data flow that is not audio so in the end may be both the global and specific features are desirable.

f4exb, 2022-07-12:
@Baa14453 Note that in this case you may also record the I/Q stream with either the FileRecord or SigMFFileRecord plugins. and demodulate the I/Q later:

  - https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/filesink
  - https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/sigmffilesink

Both have a squelch feature based on signal power. Not that the FileSink will create one file per squelch going above level while the SigMFFileSink uses SigMF format that makes a single big record with a secondary file serving as an index. This results in something more compact.

f4exb, 2022-11-12:
> Have often thought about adding it to the Demod Analyzer feature.

Yes, eventually I think having a dedicated feature just for recording would be an overkill.

f4exb, 2022-11-15:
Implemented in v7.8.3

closed 2022-11-15 by f4exb