f4exb/sdrangel #1334: dmr

F4EKN, 2022-07-10:
Bonjour,
Merci pour ce programme.
Il y a quelques années j'avais utilisé SDRANGEL pour décoder mon DM-5R et cela avait fonctionné.
J'ai changé de PC et chargé la dernière version et je suis incapable d'entendre ma modulation. Je vois bien le signal et la modulation mais je ne peux rien entendre. j'ai essayé de nombreuses combinaisons qui se sont toute terminées par un crash.
Pourriez vous regarder ma copie d'écran et me dire ou je me suis planté ?
Merci d'avance bien cordialement.
Patrick
[image]
k

f4exb, 2022-07-10:
|Firstly your receiver is overdriven as channel power shows a positive dB number (+1.4 dB) and the bright green line of the power meter reaches 0 dB. Try to lower the gain on RTL-SDR.

F4EKN, 2022-07-11:
Dear Friend,
I am not able to start AMBE because every time I select it the program closes 5 secondes after.
Can you help me ?
Thanks in advance.
Best regards

f4exb, 2022-07-12:
I don't know it works for me.

F4EKN, 2022-07-22:
Hi,
I triied many configurations and I confirm that clicking AMBE shuts down program.
Nevertheless DRM is well decode with DSDplus + SDRAngel.
73

f4exb, 2022-07-23:
You should have mentioned that it is when you check the AMBE checkbox that the problem happens. Without proper description I cannot guess what you are trying to do. 

I think this is because the AMBE feature is not present so I think this should be checked first or maybe even enable the checkbox only if an AMBE feature is present (as now we can dynamically check the presence of a feature).

f4exb, 2022-07-23:
Error is fairly straigtforward:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f142677179f in DSDDemodGUI::on_ambeSupport_clicked (this=0x564ee6a91ba0, checked=<optimized out>) at /opt/build/sdrangel/plugins/channelrx/demoddsd/dsddemodgui.cpp:276
276         m_settings.m_ambeFeatureIndex = m_availableAMBEFeatures[ui->ambeFeatures->currentIndex()].m_featureIndex;
[Current thread is 1 (Thread 0x7f14910dc840 (LWP 3456178))]
(gdb) bt
#0  0x00007f142677179f in DSDDemodGUI::on_ambeSupport_clicked(bool) (this=0x564ee6a91ba0, checked=<optimized out>) at /opt/build/sdrangel/plugins/channelrx/demoddsd/dsddemodgui.cpp:276
#1  0x00007f149557e783 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007f149612e116 in QAbstractButton::clicked(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007f149612e39e in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007f149612ffc4 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
```

f4exb, 2022-07-23:
When no AMBE feature is present the index in `ui->ambeFeatures->currentIndex()` so this is something that should be checked in the first place.

closed 2022-07-23 by f4exb

F4EKN, 2022-07-24:
Thank you for reply.
Sorry, 
that's right it's difficult to help me with so few informations.
In fact I work with Windows 10. I did not found AMBE for windows. All  programs found was distributions for linux. 
That is why I intalled DSDplus. That works but quality is quite poor.
That does not work with all devices, for example result with RTL V3 PRO is nearly good but result with HackRF is incomprehensible. 
Thank you
73

commit 97ace89e (2022-07-23): DSD demod GUI: prevent selection of negative AMBE feature index and disable AMBE checkbox if no AMBE features are available. Fixes #1334