f4exb/sdrangel #1335: Loading APRS feature cause the program to exit

as3ii, 2022-07-11:
The APRS plugin is found and loaded correctly by the plugin manager:
```
2022-07-11 14:56:08.628 (D) PluginManager::loadPluginsDir: fileName: libaprs.so
2022-07-11 14:56:08.635 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.so
```
But when I try to start it inside sdrangel the program crash and this is printed in the terminal:
```
2022-07-11 14:56:21.234 (D) MainWindow::featureAddClicked: W1 feature at 3
2022-07-11 14:56:21.234 (D) MainWindow::featureAddClicked: m_apiAdapter: 0x5bf81a07cec0
2022-07-11 14:56:21.234 (D) APRS::APRS: webAPIAdapterInterface: 0x5bf81a07cec0
2022-07-11 14:56:21.234 (D) APRS::scanAvailableChannels: register 0:3 sdrangel.channel.packetdemod (0x5bf81a55f688)
2022-07-11 14:56:21.235 (D) MessageQueueStore::createElement: 0 added
2022-07-11 14:56:21.235 (D) ObjectPipesRegistrations::registerProducerToConsumer: new pipe 0x5bf81a55f688 0x5bf81b87f0d0 MessageQueue packets
2022-07-11 14:56:21.236 (D) FeatureGUI::FeatureGUI
sdrangel: symbol lookup error: /usr/lib/sdrangel/plugins/libaprs.so: undefined symbol: _ZN8QtCharts6QChart8setThemeENS0_10ChartThemeE, version Qt_5
```
this happens regardless of the SDR.
Sdrangel installed through the `sdrangel-bin` AUR package in ArchLinux (which is based on the .deb available in the release section of this repository)

f4exb, 2022-07-12:
You are missing Qt Charts in the dependencies although it is required:
https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L408
https://github.com/f4exb/sdrangel/blob/master/.appveyor.yml#L87

I don't know how the software was built but this is wrong.

as3ii, 2022-07-12:
qt5-charts is installed as a dependency.
Maybe the problem is related to this comment? https://aur.archlinux.org/packages/sdrangel-bin#comment-856615
_"This binary and its modules are originally built for Ubuntu 20.04, as such some modules require specific library version that might not match installed version. As a consequence, these modules will not be loaded"_

srcejon, 2022-07-13:
Ubuntu 20.04 uses Qt 5.12, so not entirely surprising if it doesn't work with Qt 5.15. Your package manager should really build a new version.

closed 2022-09-18 by github-actions[bot]