f4exb/sdrangel #1339: Error while Compiling

Baa14453, 2022-07-13:
I tried compiling today following the [guide](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) I get to around 82% and it fails saying the below:

```
[ 81%] Built target modatv
Consolidate compiler generated dependencies of target aprs
[ 81%] Building CXX object plugins/channeltx/remotesource/CMakeFiles/remotesource.dir/remotesourceworker.cpp.o
[ 81%] Linking CXX shared library ../../../lib/plugins/libaprs.so
[ 82%] Built target moddatv
Consolidate compiler generated dependencies of target featuredemodanalyzer
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeaturedemodanalyzer.so
[ 82%] Built target featuregs232controller
Consolidate compiler generated dependencies of target featurepertester
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeaturepertester.so
[ 82%] Built target featureais
Consolidate compiler generated dependencies of target featurerigctlserver
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeaturerigctlserver.so
[ 82%] Built target modfreedv
[ 82%] Built target featureafc
Consolidate compiler generated dependencies of target featuresimpleptt
Consolidate compiler generated dependencies of target featuresatellitetracker
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeaturesimpleptt.so
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeaturesatellitetracker.so
Consolidate compiler generated dependencies of target jogdialcontroller
[ 82%] Linking CXX shared library ../../../lib/plugins/libjogdialcontroller.so
[ 82%] Linking CXX shared library ../../../lib/plugins/libremotesource.so
[ 82%] Built target antennatools
Consolidate compiler generated dependencies of target featureradiosonde
[ 82%] Linking CXX shared library ../../../lib/plugins/libfeatureradiosonde.so
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libavcodec.a(lpc.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/build.make:412: lib/plugins/libdemoddatv.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:6349: plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 83%] Built target featurepertester
[ 83%] Built target vorlocalizer
[ 84%] Built target featuresimpleptt
[ 84%] Built target featuredemodanalyzer
[ 84%] Built target featurerigctlserver
[ 84%] Built target jogdialcontroller
[ 84%] Built target remotesource
[ 84%] Built target featureradiosonde
[ 85%] Built target aprs
[ 86%] Built target featuresatellitetracker
[ 87%] Built target map
make: *** [Makefile:156: all] Error 2

```
I'm running Ubuntu 22, using the latest commit dcd0ee22728d41a6ec7f33bda6cd1b3c5804a857

RobRich999, 2022-07-15:
It was building on my Ubuntu 22.04 box as of a few days ago, but unless someone else has an another idea in the meantime,  you can try manually adding "-DCMAKE_POSITION_INDEPENDENT_CODE=ON" (without the quotes) to the sdrangel cmake command-line parameters.

djamic, 2022-07-16:
> It was building on my Kubuntu 22.04 box as of a few days ago, but unless someone else has an another idea in the meantime, you can try manually adding "-DCMAKE_POSITION_INDEPENDENT_CODE=ON" (without the quotes) to the sdrangel cmake command-line parameters.

I also did not find a solution to such a problem.

Baa14453, 2022-07-16:
> It was building on my Kubuntu 22.04 box as of a few days ago, but unless someone else has an another idea in the meantime, you can try manually adding "-DCMAKE_POSITION_INDEPENDENT_CODE=ON" (without the quotes) to the sdrangel cmake command-line parameters.

Sorry this hadn't fixed it for me, I've put both compile logs into files in case it helps:

[cmake.txt](https://github.com/f4exb/sdrangel/files/9125948/cmake.txt)
[make.txt](https://github.com/f4exb/sdrangel/files/9125949/make.txt)

srcejon, 2022-07-19:
The problem is you are linking against /usr/lib/x86_64-linux-gnu/libavcodec.a rather than /usr/lib/x86_64-linux-gnu/libavcodec.so

In your cmake.txt log, it says:

Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.a;/usr/lib/x86_64-linux-gnu/libavformat.so....

It should be:

Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so....

Check to see if you have /usr/lib/x86_64-linux-gnu/libavcodec.so - if not, you need to install it / maybe reinstall it.

closed 2022-09-23 by github-actions[bot]

Howie-Ye, 2024-03-28:
Replacing all 'libavcodec.a' in cmake files to 'libavcodec.so' will help.