f4exb/sdrangel #1340: Error: too many arguments to function ‘int mirisdr_open(mirisdr_dev_t**, uint32_t)'

eroyee, 2022-07-15:
Hello.

First, thanks for your work with this, and with [libmirisdr-4](https://github.com/f4exb/libmirisdr-4) which I have been trialling with my version of OpenWebRX (https://github.com/eroyee/openwebrx_E).

It is out of that work that I wanted to try SDRangel, mostly in order to see if issues I was experiencing were due to my hardware, rather than software. However I have run into an problem compiling SDRangel, ironically it would seem with mirisdr, and while I would prefer to come here with a solution I'm sorry but having spent some time with it I've been unable to resolve the error. 

Some information:

OS is Linux Mint 20.3, i7 with 32GB RAM, plenty of HDD.

After successfully installing all required dependencies and various libraries, cmake & make run ok, but make install gives the following:

```
[ 64%] Building CXX object plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/sdrplayinput.cpp.o
/media/data/l/interest/radio/develop/4/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp: In member function ‘bool SDRPlayInput::openDevice()’:
/media/data/l/interest/radio/develop/4/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:106:68: error: too many arguments to function ‘int mirisdr_open(mirisdr_dev_t**, uint32_t)’
     if ((res = mirisdr_open(&m_dev, MIRISDR_HW_SDRPLAY, m_devNumber)) < 0)
                                                                    ^
In file included from /media/data/l/interest/radio/develop/4/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.h:27,
                 from /media/data/l/interest/radio/develop/4/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:33:
/usr/local/include/mirisdr.h:59:17: note: declared here
 MIRISDR_API int mirisdr_open (mirisdr_dev_t **p, uint32_t index);

```
It's probably something I'm doing incorrectly, but a pointer as to what would be gratefully received!

Also, I saw a number of references to an Ubuntu 20.04 .deb for SDRangel, but was unable to locate the file? I expect it was right in front of me somewhere, but being late at night after a hard day at the screen...

Merci!

eroyee, 2022-07-15:
Correction, cmake runs fine, the error occurs at 'make' 🙄

ericek111, 2022-07-18:
I broke it, sowwyy >.<

nmaster2042, 2022-07-18:
Yes, this appears to be the last API change, have same error while building here.

closed 2022-07-18 by f4exb

f4exb, 2022-07-18:
Among other things the latest libmirisdr-4 brings better frequency precision.

PR #1344 (2022-07-18): SDRPlayInput: separate HW flavour from mirisdr_open
We broke SDRangel with https://github.com/f4exb/libmirisdr-4/commit/fd0452e31264f065bb315e74313384ca4a121ef1

I wasn't expecting any app to actually use the library with that argument in `mirisdr_open`.

Fixes #1340