f4exb/sdrangel #1341: DATV LDPC decoding https://github.com/f4exb/sdrangel/labels/question

DL9UFB, 2022-07-16:
It is possibele to implement LPDC in the Windows Version for receiving DATV on QO-100?

f4exb, 2022-07-16:
`ldpctool` is a Linux-centric external program. So... no.

srcejon, 2022-07-17:
> `ldpctool` is a Linux-centric external program. So... no.

Could it not be modified to build as a library? I've only had a quick glance, but it seems fairly OS independent apart from ldpc_tool.cpp, which doesn't seem that big, so could potentially be adapted.

srcejon, 2022-07-17:
With a just few changes ldpc_tool can build on Windows - the process management in leansdr/dvbs2 looks a little trickier though - but pipes to subprocesses can be made to work.

f4exb, 2022-07-17:
Or use QThread...

srcejon, 2022-07-17:
To clarify, I meant I can build ldpctool as a .exe on Windows - and in that case, pipes are needed for I/O to the child processes stdin/stdout. I've just written the code for this, but not working yet.

If it were to be rewritten as a library - then yes, QThread makes sense.

f4exb, 2022-07-18:
I see but aren't pipes used already? Cf https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddatv/leansdr/dvbs2.h#L3451

f4exb, 2022-07-20:
Implemented in v7.6.0

closed 2022-07-20 by f4exb