f4exb/sdrangel #1343: Compiler error on Macos after M17

loewal, 2022-07-18:
This is an exerpt of the nessage:

/Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/M17Demodulator.cpp:77:25: warning: lambda capture 'index' is not used [-Wunused-lambda-capture]
        correlator.apply([this,index](float t){dev.sample(t);}, index);
                              ~^~~~~
In file included from /Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/M17Demodulator.h:11:
In file included from /Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/M17FrameDecoder.h:9:
In file included from /Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/PolynomialInterleaver.h:5:
/Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/Util.h:276:32: warning: implicit conversion from 'const int' to 'signed char' changes value from 248 to -8 [-Wconstant-conversion]
    T r = v & (1 << (n - 1)) ? NEGATIVE_OFFSET : 0;
      ~                        ^~~~~~~~~~~~~~~
/Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/Viterbi.h:84:28: note: in instantiation of function template specialization 'modemm17::to_int<signed char, 4UL>' requested here
            result[i][j] = to_int<int8_t, LLR>(((bit << 1) - 1) * ((1 << (LLR - 1)) - 1));
                           ^
/Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/Viterbi.h:123:13: note: in instantiation of function template specialization 'modemm17::makeCost<modemm17::Trellis<4, 2>, 4UL>' requested here
    : cost_(makeCost<Trellis_, LLR_>(trellis))
            ^
/Users/loewal/Downloads/sdrangel-7.5.0/modems/m17/M17FrameDecoder.h:47:44: note: in instantiation of member function 'modemm17::Viterbi<modemm17::Trellis<4, 2>, 4>::Viterbi' requested here
    Viterbi<decltype(trellis_), 4> viterbi_{trellis_};
                                           ^
6 warnings and 1 error generated.
make[2]: *** [modems/CMakeFiles/modems.dir/m17/M17Demodulator.cpp.o] Error 1
make[1]: *** [modems/CMakeFiles/modems.dir/all] Error 2
make: *** [all] Error 2

On Linux there is NO error

srcejon, 2022-07-18:
@[loewal](https://github.com/loewal) - you're missing the actual error message, You might want to copy and paste the first error output.

loewal, 2022-07-18:
I'll try again...I found no other error, only lots of warnings...

In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:9:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/PolynomialInterleaver.h:5:
/Users/Loewal/Downloads/sdrangel/modems/m17/Util.h:149:29: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
    static_assert(M % N == 0);
                            ^
                            , ""
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:10:
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:17:41: error: constexpr function never produces a constant expression [-Winvalid-constexpr]
inline constexpr std::array<int8_t, 61> make_p1() {
                                        ^
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:24:13: note: non-constexpr function 'operator[]' cannot be used in a constant expression
            result[i] = 1;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/include/c++/v1/array:196:15: note: declared here
    reference operator[](size_type __n) _NOEXCEPT {
              ^
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:10:
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:110:65: warning: unused parameter 'trellis' [-Wunused-parameter]
    static constexpr cost_table_t makeCostTable(const Trellis_& trellis)
                                                                ^
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:11:
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:97:27: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
    static_assert(LLR_ < 7);    // Need to be < 7 to avoid overflow errors.
                          ^
                          , ""
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:165:49: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
        static_assert(sizeof(history_) >= IN / 2);
                                                ^
                                                , ""
/Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:77:25: warning: lambda capture 'index' is not used [-Wunused-lambda-capture]
        correlator.apply([this,index](float t){dev.sample(t);}, index);
                              ~^~~~~
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:9:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/PolynomialInterleaver.h:5:
/Users/Loewal/Downloads/sdrangel/modems/m17/Util.h:276:32: warning: implicit conversion from 'const int' to 'signed char' changes value from 248 to -8 [-Wconstant-conversion]
    T r = v & (1 << (n - 1)) ? NEGATIVE_OFFSET : 0;
      ~                        ^~~~~~~~~~~~~~~
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:84:28: note: in instantiation of function template specialization 'modemm17::to_int<signed char, 4UL>' requested here
            result[i][j] = to_int<int8_t, LLR>(((bit << 1) - 1) * ((1 << (LLR - 1)) - 1));
                           ^
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:123:13: note: in instantiation of function template specialization 'modemm17::makeCost<modemm17::Trellis<4, 2>, 4UL>' requested here
    : cost_(makeCost<Trellis_, LLR_>(trellis))
            ^
/Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:47:44: note: in instantiation of member function 'modemm17::Viterbi<modemm17::Trellis<4, 2>, 4>::Viterbi' requested here
    Viterbi<decltype(trellis_), 4> viterbi_{trellis_};
                                           ^
6 warnings and 1 error generated.
make[2]: *** [modems/CMakeFiles/modems.dir/m17/M17Demodulator.cpp.o] Error 1
make[1]: *** [modems/CMakeFiles/modems.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBeamSteeringCWModSettings.cpp.o
[  2%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1InputSettings.cpp.o
[  3%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1OutputSettings.cpp.o
[  3%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF2InputReport.cpp.o
^Cmake[2]: *** wait: Interrupted system call.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF2InputReport.cpp.o] Interrupt: 2
make[2]: *** [swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1OutputSettings.cpp.o] Interrupt: 2
make[1]: *** [swagger/CMakeFiles/swagger.dir/all] Error 2
make: *** [all] Interrupt: 2

loewal, 2022-07-18:
1 warning generated.
[  2%] Building CXX object modems/CMakeFiles/modems.dir/m17/M17Demodulator.cpp.o
[  2%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBase64Blob.cpp.o
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:9:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/PolynomialInterleaver.h:5:
/Users/Loewal/Downloads/sdrangel/modems/m17/Util.h:149:29: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
    static_assert(M % N == 0);
                            ^
                            , ""
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:10:
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:17:41: error: constexpr function never produces a constant expression [-Winvalid-constexpr]
inline constexpr std::array<int8_t, 61> make_p1() {
                                        ^
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:24:13: note: non-constexpr function 'operator[]' cannot be used in a constant expression
            result[i] = 1;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/include/c++/v1/array:196:15: note: declared here
    reference operator[](size_type __n) _NOEXCEPT {
              ^
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:10:
/Users/Loewal/Downloads/sdrangel/modems/m17/Trellis.h:110:65: warning: unused parameter 'trellis' [-Wunused-parameter]
    static constexpr cost_table_t makeCostTable(const Trellis_& trellis)
                                                                ^
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:11:
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:97:27: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
    static_assert(LLR_ < 7);    // Need to be < 7 to avoid overflow errors.
                          ^
                          , ""
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:165:49: warning: 'static_assert' with no message is a C++17 extension [-Wc++17-extensions]
        static_assert(sizeof(history_) >= IN / 2);
                                                ^
                                                , ""
/Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:77:25: warning: lambda capture 'index' is not used [-Wunused-lambda-capture]
        correlator.apply([this,index](float t){dev.sample(t);}, index);
                              ~^~~~~
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.cpp:4:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17Demodulator.h:11:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:9:
In file included from /Users/Loewal/Downloads/sdrangel/modems/m17/PolynomialInterleaver.h:5:
/Users/Loewal/Downloads/sdrangel/modems/m17/Util.h:276:32: warning: implicit conversion from 'const int' to 'signed char' changes value from 248 to -8 [-Wconstant-conversion]
    T r = v & (1 << (n - 1)) ? NEGATIVE_OFFSET : 0;
      ~                        ^~~~~~~~~~~~~~~
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:84:28: note: in instantiation of function template specialization 'modemm17::to_int<signed char, 4UL>' requested here
            result[i][j] = to_int<int8_t, LLR>(((bit << 1) - 1) * ((1 << (LLR - 1)) - 1));
                           ^
/Users/Loewal/Downloads/sdrangel/modems/m17/Viterbi.h:123:13: note: in instantiation of function template specialization 'modemm17::makeCost<modemm17::Trellis<4, 2>, 4UL>' requested here
    : cost_(makeCost<Trellis_, LLR_>(trellis))
            ^
/Users/Loewal/Downloads/sdrangel/modems/m17/M17FrameDecoder.h:47:44: note: in instantiation of member function 'modemm17::Viterbi<modemm17::Trellis<4, 2>, 4>::Viterbi' requested here
    Viterbi<decltype(trellis_), 4> viterbi_{trellis_};
                                           ^
6 warnings and 1 error generated.
make[2]: *** [modems/CMakeFiles/modems.dir/m17/M17Demodulator.cpp.o] Error 1
make[1]: *** [modems/CMakeFiles/modems.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBeamSteeringCWModSettings.cpp.o
[  2%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1InputSettings.cpp.o
[  3%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1OutputSettings.cpp.o
[  3%] Building CXX object swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF2InputReport.cpp.o
^Cmake[2]: *** wait: Interrupted system call.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF2InputReport.cpp.o] Interrupt: 2
make[2]: *** [swagger/CMakeFiles/swagger.dir/sdrangel/code/qt5/client/SWGBladeRF1OutputSettings.cpp.o] Interrupt: 2
make[1]: *** [swagger/CMakeFiles/swagger.dir/all] Error 2
make: *** [all] Interrupt: 2

loewal, 2022-07-18:
Compiled okay now Edouard! Merci et excusez mes questions stupide....

loewal, 2022-07-18:
I was too early with my optimism:
[ 54%] Building CXX object pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/sdrplayinput.cpp.o
/Users/Loewal/Downloads/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:106:16: error: no matching function for call to 'mirisdr_open'
    if ((res = mirisdr_open(&m_dev, MIRISDR_HW_SDRPLAY, m_devNumber)) < 0)
               ^~~~~~~~~~~~
/usr/local/include/mirisdr.h:59:17: note: candidate function not viable: requires 2 arguments, but 3 were provided
MIRISDR_API int mirisdr_open (mirisdr_dev_t **p, uint32_t index);
                ^
[ 54%] Linking CXX shared library ../../../lib/pluginssrv/libinputfcdproplussrv.dylib
1 error generated.
make[2]: *** [pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/sdrplayinput.cpp.o] Error 1
make[1]: *** [pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 54%] Built target inputfcdproplussrv
make: *** [all] Error 2

f4exb, 2022-07-18:
Obviously gcc on Linux does not care if you set it to c++14 so some later constructs passed through.

loewal, 2022-07-18:
My fault..installed a 'new' mirisdr and soapysdr.
Reinstalled the originals and everything was fine.!

srcejon, 2022-07-18:
Is the dll related error caused by m17_EXPORTS not being defined in export.h? It looks like its doing a dll import rather than export.

srcejon, 2022-07-18:
> So why does it work for all other "xxx_EXPORTS"?

In modems/CMakeLists.txt the project name is modems not m17

    project(modems)

So it is defining:

    -Dmodems_EXPORTS

Rather than 

    -Dm17_EXPORTS

f4exb, 2022-07-18:
> In modems/CMakeLists.txt the project name is modems not m17

Well spotted...

eroyee, 2022-07-19:
I was going to open a new issue, but this looks potentially related... After the fix in https://github.com/f4exb/sdrangel/pull/1344 (thanks), and starting again from scratch I now get the following about 1% of the way through compile:

In file included from /radio/develop/4/sdrangel/modems/m17/Golay24.cpp:2:
/radio/develop/4/sdrangel/modems/m17/Golay24.h:7:10: fatal error: bit: No such file or directory
 #include <bit>

Not sure if it's relevant but this is Mint 20.3 and cpp --version reports cpp (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

f4exb, 2022-07-19:
Yes there are two issues with the M17 library re-implementation in SDRangel:

  - Some c++17 and c++20 constructs leaked through
  - Subdirectory structure does not work with "export" one should rather have a simpler structure i.e. modems/m17/ -> modemm17/ which also matches the namespace name. Then `project(modems)` also becomes `project(modemm17)`.

alphafox02, 2022-07-19:
I didn’t have any issue compiling the latest SDRAngel on 20.04, but also didn’t pay much attention of the output while doing so. However, the end result when loading the application is that the new M17 plugin is unavailable with only an error about an undefined reference described in #1349

f4exb, 2022-07-20:
Due to too many issues the M17 plugins will be removed from the default build.

srcejon, 2022-07-20:
> Due to too many issues the M17 plugins will be removed from the default build.

I just picked up your commits earlier from today. It seems with MSVC you can get modemm17 target to compile with: 

    set(CMAKE_CXX_STANDARD 20)

In CMakeLists.txt

However, some other SDRangel code then fails:

    sdrangel\sdrgui\device\deviceuiset.cpp(233): error C2440: '<function-style-cast>': cannot convert from 'initializer list' to 'Preset::SelectedDevice'
    sdrangel\sdrgui\device\deviceuiset.cpp(238): note: No constructor could take the source type, or constructor overload resolution was ambiguous

But presumably fixable.

srcejon, 2022-07-20:
> Nothing fixes errors in MSVC that does not seem to understand anything about templates in C++ so I am giving up on this.

Worth filing an issue to the upstream project to see if they want to add support for Windows/Mac?

eroyee, 2022-07-20:
I'm happy to report the latest commits now allow me to compile. Up until today there had been a series of different issues surrounding M17.

Incidentally while Ubuntu 20.04 reports 'cpp (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0' (and I had installed gcc10) apparently it is c++14 and c++17 compatible: https://askubuntu.com/questions/1283643/how-to-update-c-or-g-to-version-14-or-above

alphafox02, 2022-07-20:
@eroyee The other ticket was closed and I realize this ticket now was not specifically about M17, but since you have it compiling  now and since I didn’t fully understand the response I was given in my ticket - can you just let me know, is the M17 plugin available or not in your successful build? My understanding is M17 is now removed.

eroyee, 2022-07-21:
@alphafox02 Please bear in mind this is the first time I've looked at the sdrangel interface (and I was really only interested in how it operated with libmirisdr-4, not any digital modes) but I assume M17 should show up directly in 'Add Features'? 

If so the then the answer is that it's not available in my build.

As far as my build is concerned I essentially followed the wiki article here: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux 

At one point I also separately compiled M17 in case that assisted with the trouble I was having, but it did not assist. Since the error I reported a couple of days ago I'd tried several times again after Edouard had made some commits around M17, all to no avail. It is only this last time that I was successful.

That's about all I can tell you, I'm happy to look again for M17 if I've not looked in the correct place! :-)

alphafox02, 2022-07-21:
Thank you for the quick turn around. I to follow that guide. I believe it should be under channels, which if you add your SDR and then to the right of the star up above the play button area, you’ll see some kind of symbol that when clicked will bring up the add channels section. My understanding according to the wiki it should be present there, which it’s not for me (I can see when starting sdrangel from terminal it’s failing). I just want to make sure I’m not crazy/doing something wrong.

eroyee, 2022-07-21:
No problem, I've had a look where you said and did find quite a range of possibilities, but no M17 (at least not directly in the top menu).

alphafox02, 2022-07-21:
Greatly appreciated, I can rest easy now haha. So either we’re both having the same issue or I’m looking in the wrong place. On a separate note, I think you’ll enjoy using all those channels one way or another (assuming you’re new to SDRAngel).

srcejon, 2022-09-12:
Now builds on Mac as well.

closed 2022-09-12 by srcejon