f4exb/sdrangel #1349: M17 tx/rx channel not available on 20.04

alphafox02, 2022-07-18:
I've tried both the deb file, doesn't seem to show any mention of M17 plugin loading and then built from source the latest SDRAngel which shows this during loading.

`2022-07-18 15:24:46.933 (W) PluginManager::loadPluginsDir: Cannot load library /opt/install/sdrangel/lib/sdrangel/plugins/libmodm17.so: (/opt/install/sdrangel/lib/sdrangel/libmodems.so: undefined symbol: _ZN8modemm1713ClockRecoveryILm48000ELm4800EE9MIN_CLOCKE)`

I'll continue to look into this.

alphafox02, 2022-07-19:
Looking over the ClockRecovery.h didn’t help, but it lead me here - which didn’t really help either https://github.com/mobilinkd/m17-cxx-demod

eroyee, 2022-07-19:
Possibly related to this https://github.com/f4exb/sdrangel/issues/1343 ? I get a different outcome (as described in that thread) but it involves M17, and I see Edouard made some recent commits with some references to M17...

alphafox02, 2022-07-19:
I see we get different results. I went and posted over in the other ticket you mentioned, although I don’t have much info to provide since the compile completed and I didn’t seem to have errors during the process. I’ll try again today.

alphafox02, 2022-07-19:
Taking a look the official 20.04 deb packages that were created after the M17 plugin reveals there's no M17 lib in sdrangel and/or plugins.

f4exb, 2022-07-20:
You don't have the FreeDV mod and demod either probably because codec2 is not properly detected. I will not fix this.

closed 2022-07-20 by f4exb

alphafox02, 2022-07-20:
Won’t fix what? Maybe I don’t understand what you mean. Your deb package tested as well doesn’t contain the files, as far I can tell. Have you tried the latest debs marked as containing m17 support?

f4exb, 2022-07-20:
Well... no. I means I will not do anything about this.

alphafox02, 2022-07-20:
I’m not following you then, are you saying I’ve done something wrong and you’re not looking to fix what I’ve done wrong or are you saying m17 just isn’t working right in SDRAngel and that you’re not doing anything about that except to remove it for right now? Wish I could help if it’s the latter.

alphafox02, 2022-07-23:
Circling back around to this after hearing from at least one other person that compiling SDRAngel from source does not present the m17 plugin. 

Can you please clarify weather or not the plug-in is building and usable on some Linux distros? Your responses are rather short and I can’t tell if you’re only speaking to the deb package not having the plug-in or if your saying that even building from source will not allow it to work right now. 

@srcejon maybe you can expand on this too so I don’t keep chasing something if it’s not possible right now. I thought about going to a 22.04 machine and seeing if I get the same issue.

alphafox02, 2022-07-23:
> You don't have the FreeDV mod and demod either probably because codec2 is not properly detected. I will not fix this.

Freedv modulator and demod  is actually present.

alphafox02, 2022-07-23:
I just created a different ticket since there seemed to be some confusion on this one. Hopefully the second is cleaner and offers a better understanding of the problem.