f4exb/sdrangel #1353: Compile error in sdrplayinput.cpp

lrowen, 2022-07-19:
I've been following sdrangel for some time now, successfully going through v7, building from source.  

This morning, ran into a compilation error with the latest git pull:

`[ 53%] Building CXX object pluginssrv/samplesource/sdrplayv3/CMakeFiles/inputsdrplayv3srv.dir/sdrplayv3input.cpp.o
/opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp: In member function ‘bool SDRPlayInput::openDevice()’:
/opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:106:37: error: invalid conversion from ‘int’ to ‘mirisdr_hw_flavour_t’ [-fpermissive]
  106 |     if ((res = mirisdr_open(&m_dev, m_devNumber)) < 0)
      |                                     ^~~~~~~~~~~
      |                                     |
      |                                     int
/opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:106:48: error: too few arguments to function ‘int mirisdr_open(mirisdr_dev_t**, mirisdr_hw_flavour_t, uint32_t)’
  106 |     if ((res = mirisdr_open(&m_dev, m_devNumber)) < 0)
      |                                                ^
In file included from /opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.h:27,
                 from /opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:33:
/opt/install/libmirisdr/include/mirisdr.h:59:17: note: declared here
   59 | MIRISDR_API int mirisdr_open (mirisdr_dev_t **p, mirisdr_hw_flavour_t hw_flavour, uint32_t index);
      |                 ^~~~~~~~~~~~
/opt/build/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:112:16: error: ‘mirisdr_set_hw_flavour’ was not declared in this scope; did you mean ‘mirisdr_hw_flavour_t’?
  112 |     if ((res = mirisdr_set_hw_flavour(m_dev, MIRISDR_HW_SDRPLAY)) < 0)
      |                ^~~~~~~~~~~~~~~~~~~~~~
      |                mirisdr_hw_flavour_t
[ 53%] Building CXX object pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/sdrplaysettings.cpp.o
[ 53%] Building CXX object pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/sdrplaywebapiadapter.cpp.o
make[2]: *** [pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/build.make:95: pluginssrv/samplesource/sdrplay/CMakeFiles/inputsdrplaysrv.dir/sdrplayinput.cpp.o] Error 1
`
Git information:
`lowen@lowen-7740:/opt/build/sdrangel$ git show -s
commit 2612d6793edb16cdd172fe29e31c49e4817e1357 (HEAD -> master, origin/master, origin/HEAD)
Author: f4exb <f4exb06@gmail.com>
Date:   Tue Jul 19 17:42:24 2022 +0200

    DATV demod: use non blocking read in ldpctool to fix issue in PR #1345
lowen@lowen-7740:/opt/build/sdrangel$ 
`

As of yesterday it built without error.

srcejon, 2022-07-19:
> EDIT: Do I need a newer mirisdr build?

Quite possibly: https://github.com/f4exb/sdrangel/pull/1344

lrowen, 2022-07-19:
Ok, updated libmirisdr-4 to:
``` lowen@lowen-7740:/opt/build/libmirisdr-4$ git show -s
commit fd0452e31264f065bb315e74313384ca4a121ef1 (HEAD -> master, tag: v2.0.0, origin/master, origin/HEAD)
Merge: 4c7f505 e9cd752
Author: Edouard Griffiths <f4exb06@gmail.com>
Date:   Wed Apr 27 19:02:36 2022 +0200

    Merge pull request #9 from ericek111/master
    
    move hw_flavour from the initializer into a separate API function
lowen@lowen-7740:/opt/build/libmirisdr-4$ 
```
That seems to have fixed it.

closed 2022-07-19 by lrowen