f4exb/sdrangel #1359: Build fails on ARM64

AsciiWolf, 2022-07-21:
Trying to build SDRangel v7.6.0 on a 64-bit ARM results in this error:

```
[1537/2590] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/glshaderspectrogram.cpp.o
FAILED: sdrgui/CMakeFiles/sdrgui.dir/gui/glshaderspectrogram.cpp.o 
/usr/bin/c++ -DAPPLICATION_NAME=\"SDRangel\" -DARCHITECTURE_ARM64 -DCOMPANY=\"f4exb\" -DCOPYRIGHT="\"Copyright (c) 2019 Edouard Griffiths. All rights reserved.\"" -DHAS_LIMERFEUSB -DIDENTIFIER=\"com.f4exb.SDRangel\" -DLIB=\"lib\" -DLINUX -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DQT_OPENGL_LIB -DQT_WEBSOCKETS_LIB -DQT_WIDGETS_LIB -DSDRANGEL_LIB_VERSION=\"7.6.0\" -DSDRANGEL_VERSION=\"7.6.0\" -DSDR_RX_SAMPLE_24BIT -DUSE_NEON -Dsdrgui_EXPORTS -I/run/build/sdrangel/sdrgui -I/run/build/sdrangel/sdrgui/sdrgui_autogen/include -I/run/build/sdrangel/exports -I/run/build/sdrangel/sdrbase -I/run/build/sdrangel/logging -I/run/build/sdrangel/httpserver -I/run/build/sdrangel/swagger/sdrangel/code/qt5/client -isystem /usr/include/QtCore -isystem /usr/lib/mkspecs/linux-g++ -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtOpenGL -isystem /usr/include/QtMultimedia -isystem /usr/include/QtNetwork -isystem /usr/include/QtWebSockets -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -O2 -g -DNDEBUG -flto -fno-fat-lto-objects -fPIC -march=native -Wall -Wextra -Wvla -Woverloaded-virtual -ffast-math -fno-finite-math-only -ftree-vectorize -fPIC -std=c++14 -MD -MT sdrgui/CMakeFiles/sdrgui.dir/gui/glshaderspectrogram.cpp.o -MF sdrgui/CMakeFiles/sdrgui.dir/gui/glshaderspectrogram.cpp.o.d -o sdrgui/CMakeFiles/sdrgui.dir/gui/glshaderspectrogram.cpp.o -c /run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp: In member function ‘void GLShaderSpectrogram::initColorMapTextureImmutable(const QString&)’:
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp:254:35: warning: ‘void QOpenGLTexture::setData(QOpenGLTexture::PixelFormat, QOpenGLTexture::PixelType, void*, const QOpenGLPixelTransferOptions*)’ is deprecated [-Wdeprecated-declarations]
  254 |         m_colorMapTexture->setData(QOpenGLTexture::RGB, QOpenGLTexture::Float32, colorMap);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/QtGui/QtGui:57,
                 from /usr/include/QtOpenGL/QtOpenGLDepends:4,
                 from /usr/include/QtOpenGL/QtOpenGL:3,
                 from /run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp:23:
/usr/include/QtGui/qopengltexture.h:469:24: note: declared here
  469 |     QT_DEPRECATED void setData(PixelFormat sourceFormat, PixelType sourceType,
      |                        ^~~~~~~
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp: In member function ‘void GLShaderSpectrogram::initColorMapTextureMutable(const QString&)’:
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp:269:19: error: ‘GL_TEXTURE_1D’ was not declared in this scope; did you mean ‘GL_TEXTURE_2D’?
  269 |     glBindTexture(GL_TEXTURE_1D, m_colorMapTextureId);
      |                   ^~~~~~~~~~~~~
      |                   GL_TEXTURE_2D
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp:272:9: error: ‘glTexImage1D’ was not declared in this scope; did you mean ‘glTexImage2D’?
  272 |         glTexImage1D(GL_TEXTURE_1D, 0, GL_RGB, 256, 0, GL_RGB, GL_FLOAT, colorMap);
      |         ^~~~~~~~~~~~
      |         glTexImage2D
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp: In member function ‘void GLShaderSpectrogram::drawSurface(SpectrumSettings::SpectrogramStyle, const QMatrix4x4&, float, bool)’:
/run/build/sdrangel/sdrgui/gui/glshaderspectrogram.cpp:415:23: error: ‘GL_TEXTURE_1D’ was not declared in this scope; did you mean ‘GL_TEXTURE_2D’?
  415 |         glBindTexture(GL_TEXTURE_1D, m_colorMapTextureId);
      |                       ^~~~~~~~~~~~~
      |                       GL_TEXTURE_2D
```

srcejon, 2022-07-21:
Looks like no GL_TEXTURE_1D in Open GL ES 2. 

Might need to change to a 2D texture of height 1.

srcejon, 2022-08-02:
Can you let me know if the above patch has helped? I can't build for ARM.

AsciiWolf, 2022-08-02:
I did not try the patch, I wanted to wait for the next SDRangel release.

CRD716, 2022-08-19:
> I did not try the patch, I wanted to wait for the next SDRangel release.

Does the newest maintenance release work?

AsciiWolf, 2022-08-19:
Sorry for not updating the ticket. It indeed does work. :-)

closed 2022-08-19 by AsciiWolf