f4exb/sdrangel #1362: ISM demodulator

AsciiWolf, 2022-07-21:
It would be great if SDRangel an ISM band data receiver similar to [rtl_433](https://github.com/merbanan/rtl_433). :-)

srcejon, 2022-07-22:
Rather than "ISM decoder", I'd probably say we should work towards having a generic digital demodulator, supporting user-selectable modulation, data-rate, clock-recovery and so on, with some extensible mechanism for decoding the packets in to display form (maybe based on something like ASN.1).

Something I want to do, but need to improve some of the basics in the existing digital demods first.

AsciiWolf, 2022-09-23:
I agree. The "generic demodulator" could also be useful for demodulating rtty/psk31 signals, pocsag and other paging protocols and possibly also ft8/jt9 and similar.

srcejon, 2022-09-23:
There's already a POCSAG demod: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodpager/readme.md

Note that you can only do so much with a generic decoder. E.g. the POCSAG demod supports error correction, and there's a few other protocol specific bits in there, that you probably wouldn't have in a generic demod.

AsciiWolf, 2022-09-23:
Ah, I forgot about the POCSAG demod. :-)