f4exb/sdrangel #1365: 3d map disappears AIS demodulator bug ??

choupette-change, 2022-07-26:
After a while (15-20 minutes or more) the 3d map disappears??
[image]

srcejon, 2022-07-27:
Are there any messages in the log file?

choupette-change, 2022-07-27:
the problem just happened again
[image]
[sdrangel.log](https://github.com/f4exb/sdrangel/files/9199360/sdrangel.log)

choupette-change, 2022-07-27:
another case
[sdrangel.log](https://github.com/f4exb/sdrangel/files/9201361/sdrangel.log)

srcejon, 2022-09-27:
More info here:

https://github.com/f4exb/sdrangel/issues/1423

What graphics card do you have and how much memory does it have?

choupette-change, 2022-09-27:
Hi Jon,
tutu@tutu-Aspire-E5-573:~$ lspci -vnn | grep -A 12 '\''[030[02]\]' | grep -Ei "vga|3d|display|kernel"
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 5500 [8086:1616] (rev 09) (prog-if 00 [VGA controller])
	Kernel driver in use: i915
	Kernel modules: i915
tutu@tutu-Aspire-E5-573:~$ 
tutu@tutu-Aspire-E5-573:~$ sudo lshw -c video
[sudo] Mot de passe de tutu : 
  *-display                 
       description: VGA compatible controller
       produit: HD Graphics 5500
       fabricant: Intel Corporation
       identifiant matériel: 2
       information bus: pci@0000:00:02.0
       nom logique: /dev/fb0
       version: 09
       bits: 64 bits
       horloge: 33MHz
       fonctionnalités: msi pm vga_controller bus_master cap_list rom fb
       configuration : depth=32 driver=i915 latency=0 resolution=1366,768
       ressources : irq:50 mémoire:c0000000-c0ffffff mémoire:b0000000-bfffffff portE/S:4000(taille=64) mémoire:c0000-dffff
Cheers

choupette-change, 2022-09-27:
utu@tutu-Aspire-E5-573:~$ lspci -vnn | grep -A 12 '\''[030[02]\]' | grep -Ei "vga|3d|display|kernel"
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 5500 [8086:1616] (rev 09) (prog-if 00 [VGA controller])
	Kernel driver in use: i915
	Kernel modules: i915
tutu@tutu-Aspire-E5-573:~$ sudo lshw -c video
[sudo] Mot de passe de tutu : 
  *-display                 
       description: VGA compatible controller
       produit: HD Graphics 5500
       fabricant: Intel Corporation
       identifiant matériel: 2
       information bus: pci@0000:00:02.0
       nom logique: /dev/fb0
       version: 09
       bits: 64 bits
       horloge: 33MHz
       fonctionnalités: msi pm vga_controller bus_master cap_list rom fb
       configuration : depth=32 driver=i915 latency=0 resolution=1366,768
       ressources : irq:50 mémoire:c0000000-c0ffffff mémoire:b0000000-bfffffff portE/S:4000(taille=64) mémoire:c0000-dffff
tutu@tutu-Aspire-E5-573:~$

srcejon, 2022-10-15:
Try replacing some of the larger 3D models, with a smaller one:

cd ~/.local/share/f4exb/SDRangel/3d

cp tug_20m.glbe tanker_245m_1.glbe
cp tug_20m.glbe tanker_245m_2.glbe
cp tug_20m.glbe tanker_245m_3.glbe
cp tug_20m.glbe tanker_380m_1.glbe
cp tug_20m.glbe tanker_380m_2.glbe
cp tug_20m.glbe tanker_50m.glbe

closed 2022-12-21 by github-actions[bot]