f4exb/sdrangel #1369: sdrangel hangs after any map attempts to load

CRD716, 2022-08-02:
Version 7.6.0-42-g75148f965
Built from source on Debian (Sid)
Qt 5.15.4 64bit Rx 24bit Tx 16bit
HackRF One (Latest firmware)

After trying to use either the map feature or the ADS-B channel, sdrangel hangs (seemingly indefinitely) 

sdrangel.log:
```
2022-08-02 15:30:22.800 (I) SDRangel 7.6.0-42-g75148f965 Qt 5.15.4 64b x86_64 Debian GNU/Linux bookworm/sid DSP Rx:24b Tx:16b PID 1271
2022-08-02 15:30:32.277 (D) ADSBDemod::ADSBDemod
2022-08-02 15:30:32.278 (D) ADSBDemodSink::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_correlationThreshold:  10  m_correlateFullPreamble:  true  m_demodModeS:  false  m_samplesPerBit:  4  force:  true
2022-08-02 15:30:32.278 (D) Interpolator::createPolyphaseLowPass: ntaps: 14
2022-08-02 15:30:32.279 (D) ADSBDemodSink::applyChannelSettings:  channelSampleRate:  6000000  channelFrequencyOffset:  0
2022-08-02 15:30:32.279 (D) NCO freq: 0.000000 phase inc 0
2022-08-02 15:30:32.279 (D) Interpolator::createPolyphaseLowPass: ntaps: 14
2022-08-02 15:30:32.309 (D) ADSBDemodBaseband::ADSBDemodBaseband
2022-08-02 15:30:32.309 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2022-08-02 15:30:32.310 (D) DSPDeviceSourceEngine::addSink:  ADSBDemod
2022-08-02 15:30:32.310 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2022-08-02 15:30:32.312 (D) ChannelGUI::ChannelGUI
2022-08-02 15:30:32.392 (D) AircraftInformation::readFastDB:  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabaseFast.csv"
2022-08-02 15:30:32.392 (D) AircraftInformation::readFastDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabaseFast.csv"
2022-08-02 15:30:32.392 (D) AircraftInformation::readFastDB - read  0  aircraft
2022-08-02 15:30:32.392 (D) AircraftInformation::readOSNDB:  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabase.csv"
2022-08-02 15:30:32.392 (D) AircraftInformation::readOSNDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabase.csv"
2022-08-02 15:30:32.392 (D) AircraftInformation::readOSNDB: Read  0  aircraft
2022-08-02 15:30:32.392 (D) AirportInformation::readAirportsDB:  "/home/crd716/.local/share/f4exb/SDRangel/airportDatabase.csv"
2022-08-02 15:30:32.392 (D) AirportInformation::readAirportsDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/airportDatabase.csv"
2022-08-02 15:30:32.392 (D) AirportInformation::readAirportsDB: Read  0  airports
2022-08-02 15:30:32.392 (D) csvHash:  ":/flags/regprefixmap.csv"
2022-08-02 15:30:32.392 (D) csvHash:  ":/flags/regprefixmap.csv" : read  223  entries
2022-08-02 15:30:32.392 (D) csvHash:  ":/flags/militarymap.csv"
2022-08-02 15:30:32.393 (D) csvHash:  ":/flags/militarymap.csv" : read  47  entries
2022-08-02 15:30:35.160 (D) ADSBDemodGUI::updateDeviceSetLists: device index changed: 0
2022-08-02 15:30:35.178 (D) ADSBDemodGUI::applySettings
2022-08-02 15:30:35.178 (D) ADSBDemod::handleMessage: DSPSignalNotification
2022-08-02 15:30:35.179 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:30:35.179 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2022-08-02 15:30:35.179 (D) MainWindow::channelAddClicked: adding ADS-B Demodulator to workspace #0
2022-08-02 15:30:35.202 (D) ADSBDemodGUI::applySettings
2022-08-02 15:30:35.202 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:30:35.203 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2022-08-02 15:30:35.225 (D) ADSBDemodGUI::applySettings
2022-08-02 15:30:35.225 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:30:35.225 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
```

GDB output:
```
2022-08-02 15:45:32.462 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2022-08-02 15:45:32.462 (D) DSPDeviceSourceEngine::addSink:  ADSBDemod
2022-08-02 15:45:32.462 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2022-08-02 15:45:32.465 (D) ChannelGUI::ChannelGUI
[New Thread 0x7fff78b20640 (LWP 2480)]
2022-08-02 15:45:32.737 (D) AircraftInformation::readFastDB:  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabaseFast.csv"
2022-08-02 15:45:32.737 (D) AircraftInformation::readFastDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabaseFast.csv"
2022-08-02 15:45:32.737 (D) AircraftInformation::readFastDB - read  0  aircraft
2022-08-02 15:45:32.737 (D) AircraftInformation::readOSNDB:  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabase.csv"
2022-08-02 15:45:32.737 (D) AircraftInformation::readOSNDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/aircraftDatabase.csv"
2022-08-02 15:45:32.737 (D) AircraftInformation::readOSNDB: Read  0  aircraft
2022-08-02 15:45:32.737 (D) AirportInformation::readAirportsDB:  "/home/crd716/.local/share/f4exb/SDRangel/airportDatabase.csv"
2022-08-02 15:45:32.737 (D) AirportInformation::readAirportsDB: Failed to open  "/home/crd716/.local/share/f4exb/SDRangel/airportDatabase.csv"
2022-08-02 15:45:32.737 (D) AirportInformation::readAirportsDB: Read  0  airports
2022-08-02 15:45:32.737 (D) csvHash:  ":/flags/regprefixmap.csv"
2022-08-02 15:45:32.737 (D) csvHash:  ":/flags/regprefixmap.csv" : read  223  entries
2022-08-02 15:45:32.737 (D) csvHash:  ":/flags/militarymap.csv"
2022-08-02 15:45:32.737 (D) csvHash:  ":/flags/militarymap.csv" : read  47  entries
[Detaching after fork from child process 2481]
[New Thread 0x7fff614be640 (LWP 2502)]
2022-08-02 15:45:35.405 (D) ADSBDemodGUI::updateDeviceSetLists: device index changed: 0
[New Thread 0x7fff60cbd640 (LWP 2507)]
2022-08-02 15:45:35.434 (D) ADSBDemodGUI::applySettings
2022-08-02 15:45:35.434 (D) ADSBDemod::handleMessage: DSPSignalNotification
2022-08-02 15:45:35.434 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:45:35.434 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2022-08-02 15:45:35.434 (D) MainWindow::channelAddClicked: adding ADS-B Demodulator to workspace #0
2022-08-02 15:45:35.450 (D) ADSBDemodGUI::applySettings
2022-08-02 15:45:35.450 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:45:35.450 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
[New Thread 0x7fff558ee640 (LWP 2508)]
2022-08-02 15:45:35.472 (D) ADSBDemodGUI::applySettings
2022-08-02 15:45:35.472 (D) ADSBDemod::handleMessage: MsgConfigureADSBDemod
2022-08-02 15:45:35.472 (D) ADSBDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  2.6e+06  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2022-08-02 15:45:35.508 (W) QGeoTileFetcherOsm: all providers resolved
[New Thread 0x7fff549ed640 (LWP 2509)]
[New Thread 0x7fff47fff640 (LWP 2510)]
[Thread 0x7fffa23fd640 (LWP 2316) exited]
[Thread 0x7fffa2bfe640 (LWP 2315) exited]
[Thread 0x7fff9ebfe640 (LWP 2314) exited]
[Thread 0x7fffa33ff640 (LWP 2313) exited]
[Thread 0x7fff47fff640 (LWP 2510) exited]
```

srcejon, 2022-08-02:
There is a Qt 5.15.3 Map bug when OSM maps are used: https://bugreports.qt.io/browse/QTBUG-103592

I presume the same problem exists in 5.15.4, that you appear to have, and unfortunately the workaround in this patch https://github.com/f4exb/sdrangel/issues/1169 only checks for 5.15.3:

    #if QT_VERSION == QT_VERSION_CHECK(5, 15, 3)
        m_mapProvider = "mapboxgl"; // osm maps do not work in Qt 5.15.3 - https://github.com/f4exb/sdrangel/issues/1169
    #else
        m_mapProvider = "osm";
    #endif

CRD716, 2022-08-02:
Changing that one line fixed it for me. Works like a charm.

srcejon, 2022-08-02:
Reopened as I need to fix it for others

jtaylor2, 2022-09-27:
Qt 5.15.2 also has this bug.  I  upgraded my openSUSE  from LEAP 15.3 to 15.4 (latest) so I could continue to compile this project after the Qt required version changed and hit this issue (openSUSE 15.4 supplies Qt version 5.15.2).

I have patch my local system but it would be nice if someone with local repository access could add Qt 5.15.2 to the two version checks.  If not I can fork it and do a pull request when I get some time.
```
diff --git a/plugins/feature/map/mapsettings.cpp b/plugins/feature/map/mapsettings.cpp
index b9eaff565..99b5e99dd 100644
--- a/plugins/feature/map/mapsettings.cpp
+++ b/plugins/feature/map/mapsettings.cpp
@@ -88,7 +88,7 @@ MapSettings::~MapSettings()
 void MapSettings::resetToDefaults()
 {
     m_displayNames = true;
-#if QT_VERSION == QT_VERSION_CHECK(5, 15, 3) || QT_VERSION == QT_VERSION_CHECK(5, 15, 4)
+#if QT_VERSION == QT_VERSION_CHECK(5, 15, 3) || QT_VERSION == QT_VERSION_CHECK(5, 15, 4) || QT_VERSION_CHECK(5, 15, 2)
     m_mapProvider = "mapboxgl"; // osm maps do not work in Qt 5.15.2/3/4 - https://github.com/f4exb/sdrangel/issues/1169
 #else
     m_mapProvider = "osm";
~

diff --git a/plugins/channelrx/demodadsb/adsbdemodsettings.cpp b/plugins/channelrx/demodadsb/adsbdemodsettings.cpp
index 5561296c3..50d2c3cd9 100644
--- a/plugins/channelrx/demodadsb/adsbdemodsettings.cpp
+++ b/plugins/channelrx/demodadsb/adsbdemodsettings.cpp
@@ -92,7 +92,7 @@ void ADSBDemodSettings::resetToDefaults()
     m_logEnabled = false;
     m_airspaces = QStringList({"A", "D", "TMZ"});
     m_airspaceRange = 500.0f;
-#if QT_VERSION == QT_VERSION_CHECK(5, 15, 3) || QT_VERSION == QT_VERSION_CHECK(5, 15, 4)
+#if QT_VERSION == QT_VERSION_CHECK(5, 15, 3) || QT_VERSION == QT_VERSION_CHECK(5, 15, 4) || QT_VERSION_CHECK(5, 15,2)
     m_mapProvider = "mapboxgl"; // osm maps do not work in Qt 5.15.3 - https://github.com/f4exb/sdrangel/issues/1169
 #else
     m_mapProvider = "osm";
```

srcejon, 2022-09-27:
Well, it's not as simple as saying 5.15.2 has the problem, as this version is fine on Windows and Mac, and also Ubuntu 21.10 has Qt 5.15.2 and doesn't show the problem.

jtaylor2, 2022-09-27:
Well that's interesting that Ubuntu's supplied Qt5.15.2 doesn't have the problem.  I wonder what the difference is between there version and openSUSE's ?  I will have to do some research.
openSUSE Leap 15.4 supplies:
```
Name           : libQt5Core-devel
Version        : 5.15.2+kde294-150400.6.3.1
```

srcejon, 2022-09-30:
Qt 5.15.6 has the problem on Ubuntu 22.10 beta

CRD716, 2022-09-30:
Updating name as we have known about the cause for a while now

PR #1370 (2022-08-03): Fix QT 5.15.4 map hang
Closes #1369
closed 2022-10-01 by srcejon