f4exb/sdrangel #1371: Crashes with "Not Reposnding" when it attempts to launch on non-primary monitor. Windows 10

adammelancon, 2022-08-03:
I finally got sdrangel to launch by disabling my 3 other monitors and having sdrangel start on my primary main monitor. 
It then ran fine.

Here is the issue.
When starting versions 7.4 to 7.61, on my Windows 10 Pro Desktop, the initial logo will pop up showing it loading, but when the app starts, it wants to launch on my left monitor (I did previously have a much older version installed).  When it attempts to start, it is just a white window that after a few seconds will say "Not Responding".
I have tried multiple versions of sdrangel from 7.4-7.61 and and removed and reinstalled all of the C++ runtime versions.  I have also tried with the nesdr from nooelec plugged in and unplugged.
The only thing that instantly fixed the issue was disabling all of my other monitors, then restarting sdrangel to have it launch on my primary monitor.

I then closed sdrangel and re-enabled my other monitors. I then started sdrangel and it launched in my primary monitor and had no issues.

f4exb, 2022-08-04:
In the context that is saved at exit time the placement of the SDRagnel window is remembered so when you start again it will start on the primary or the secondary (tertiary etc...) display depending on where it was left at the last exit. The splash screen on the other hand will always show on the primary display. However I still cannot find any issue with this behavior.

adammelancon, 2022-08-04:
The behavior is perfect with it, as far as it resuming on the last monitor it was used.  That's a great feature.
In this case, it was just locking up completely when the last session was saved on another monitor.  Once I disabled all monitors except my main monitor, it finally started up correctly on that screen.
If it tried resuming on my 2nd monitor, the application would stay white and say "not responding" at the top.
Perhaps this stems from my having a much older version installed and not having launched it in some time, then installing the newest version, and launching the new version while it tried to maintain the previous window location.  I'm not sure, but I'm glad I did get this amazing program to work again!  Just wanted to report it in case someone else ran into the same issue.

closed 2022-10-09 by github-actions[bot]