f4exb/sdrangel #1372: Unfortunately 7.6.1 crashes on close under Windows.

mistergitj, 2022-08-03:
Thanks for the latest version (7.6.1) of SDRangel.  I am running under W10.  Unfortunately it crashes when I click the X button (Close it).  Enjoy, John.

srcejon, 2022-08-04:
Works OK for me.

What SDR are you using? What plugins/features do you have open before pressing close? A screenshot would be useful.

f4exb, 2022-08-04:
@mistergitj indeed your description is too vague and discourages me to continue developing this software

mistergitj, 2022-08-04:
Sorry for the vague post.  I erroneously thought that my first reporting this problem, [here](https://github.com/f4exb/sdrangel/issues/1312#issuecomment-1171906278), would be recalled.
I am running the AirSpy R2 and it does not seem to matter what configuration I am running.  If I open SDRangel, load any configuration, make changes and close SDRangel, all is fine.  If I Start the configuration, it always crashes on close after clicking the X on the main dialog or X on the SDRangel window. Thanks and enjoy, John.

mistergitj, 2022-08-05:
A friend loaned me an SDRplay and it does crash on W10 using v7.6.1. Hope this helps. Enjoy, John.

mistergitj, 2022-10-05:
I have continued to test this problem on each new release and the results are the same. Using W10 with an Airspy R2,  if I start the radio, SDRangel crashes on close using the X.  If I have modified a Configuration, the changes are lost. Either I am doing something wrong or SDRangel has a problem with W10.  My SDRplay never crashes on close. Thanks, John.

srcejon, 2022-10-05:
I don't have an airspy, so can't test this myself. However, a few changes have been made to the airspy driver regarding threading, so perhaps that might make a difference. If you want to try them, download this latest test build:

https://github.com/f4exb/sdrangel/suites/8588595921/artifacts/384751205

mistergitj, 2022-10-05:
Thanks, srcejon, I ran a quick test on the sdrangel-7.7.0-102-ga94752602-win64 but it crashed as soon as I clicked the X in the device window.  I get a blue circle when I click. I am running W10 x64. In the original thread I opened, [here](https://github.com/f4exb/sdrangel/issues/1312#issuecomment-1171906278), this problem was reproduced on your end. Thanks for your response, John.

srcejon, 2022-10-05:
Ok, here's another attempt at a fix: 

https://github.com/f4exb/sdrangel/suites/8628210569/artifacts/387514948

mistergitj, 2022-10-05:
Thanks, srcejon, will test after lunch. Enjoy, John.

mistergitj, 2022-10-05:
Unfortunately, srcejon but got the same results.  Here is what showed up in the Windows Event Log:
- System 

  - Provider 

   [ Name]  Application Error 
 
  - EventID 1000 

   [ Qualifiers]  0 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x80000000000000 
 
  - TimeCreated 

   [ SystemTime]  2022-10-05T18:16:55.1242372Z 
 
   EventRecordID 37763 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  0 
   [ ThreadID]  0 
 
   Channel Application 
 
   Computer xxxxxx 
 
   Security 
 

- EventData 

   sdrangel.exe 
   0.0.0.0 
   633d9f2b 
   libusb-1.0.dll 
   1.0.26.11724 
   6252cb06 
   c0000005 
   0000000000001d05 
   338c 
   01d8d8e64918014e 
   C:\Program Files\SDRangel\sdrangel.exe 
   C:\Program Files\SDRangel\libusb-1.0.dll 
   e011efcf-8d51-4d16-b272-7d6e1bfb8839 
    
    Thanks, John.

srcejon, 2022-10-07:
So I've managed to reproduce the problem. It appears to be related to the version of libusb used. With 1.0.26, it crashes, with 1.0.25 it doesn't.

[image]

Not obvious what sdrangel could be doing wrong here, as this is when it calls airspyhf_close().

mistergitj, 2022-10-07:
Great! Thanks, srcejon. Thanks for your work.  Enjoy, john.

srcejon, 2022-10-07:
Ok, my crash was with AirspyHF+ rather than Airspy, and is due to a bug in the airspyhf driver which was fixed here:

https://github.com/airspy/airspyhf/commit/c55a305a79eea70cf61ea22a081b29d37805b9fc

So we need to update AIRSPYHF_TAG to a more recent tag.

It looks like the same bug was fixed in the Airspy driver, so we also need to update that:

https://github.com/airspy/airspyone_host/commit/41c439f16818d931c4d0f8a620413ea5131c0bd6

closed 2022-10-07 by f4exb

f4exb, 2022-10-07:
I have a Airspy mini and it does crash on close with the previous revision. Let's wait for the build...

mistergitj, 2022-10-19:
Thanks much!  I will some more testing tomorrow but a quick test showed 7.8.1 does no crash on close using an AirSpy under W10.  Thank you very much, John.

PR #1468 (2022-10-07): Update to latest airspy drivers to fix #1372
Update to latest airspy drivers to fix #1372

Note, I don't have an Airspy, so can't test to see if there are any problems. AirspyHF+ seems to work ok.
commit 2bdc456b (2022-10-07): Merge pull request #1468 from srcejon/fix_1372

Update to latest airspy drivers to fix #1372
commit 85bf44c1 (2022-10-07): Update to latest airspy drivers to fix #1372