f4exb/sdrangel #1373: Installer relies on ubuntu virtual package qtbase-abi-5-12-8, which is not available on debian

shifter2015, 2022-08-04:
`qtbase-abi-5-12-8` is a virtual package used in Ubuntu which is provided by `libqt5core5a`, however this `qtbase-abi-5-12-8` is not available on Debian 11. This can be circumvented by using `equivs` to generate a fake package to satisfy the dependencies, but it would be nice if the installer just looked for `libqt5core5a` instead of `qtbase-abi-5-12-8`.

f4exb, 2022-08-04:
Are you trying to install the provided .deb package on Debian 11? It is well documented that this is intended for Ubuntu 20.04. In all other cases you should build the code from source.

shifter2015, 2022-08-04:
> 

Ah, i see its alright then, its an easy issue to circumvent anyway.

closed 2022-08-04 by shifter2015

knowaveragejoe, 2022-08-15:
Can confirm I was able to bypass this issue trivially using `equivs` in a Debian-based system.