f4exb/sdrangel #1376: NFM demod is not working

darknebular, 2022-08-05:
Hello, I can demodulate NFM perfectly in SDR#, when I go to SDRAngel, set the same frecuency and select NFM Demod, I can´t hear anything, all gains, frecuencies, are the same with SDR# but It looks like that is not working.
If I turn off the sq option, I only hear white noise.
I have HackRF, same antenna, same USB cable, same location and same PC.

With the same ganancies, same bandwitdh and same Frecuency in SDR# I am having excelent results with SDRAngel and NFM demod not, only white noise, or nothing.

Best regards.

srcejon, 2022-08-05:
A screenshot would help.

f4exb, 2022-08-05:
Just use SDR#

darknebular, 2022-08-08:
> Just use SDR#

SDR# hasn´t got transmition. I would like to use SDRAngel with PTT for NFM.

darknebular, 2022-08-08:
> A screenshot would help.

It´s about sound not image...

srcejon, 2022-08-08:
> > A screenshot would help.
> 
> It´s about sound not image...

So it would be useful to see your settings and signal in an image.

darknebular, 2022-08-08:
[image]
I can hear fine
[image]
I only listen noise o very very awful voice.

Best regards.

srcejon, 2022-08-08:
It's a bit hard to see the Spectrum - can I suggest you press the "AS" button in the toolbar so that the windows are automatically resized.

I would guess that the problem may be related to you having DC offset removal checked in HackRF settings (The DC button that is orange), and your NFM Demodulator has a frequency offset of 0 (Delta F is 0 - first setting in top left of NFM Demod). This probably means that you are filtering the signal you are trying to listen to. 

I'm not familar with HackRF, but it's generally best not to have your demodulators exactly at the device centre frequency, because of the DC offset most radios generate. So, try setting Device Centre Frequency to 145,300kHz then put the NFM Demodulator at an offset of 37,000 Hz.

Then also try to zoom in a little on the spectrum so we can see the signal a bit more clearly. You can start by pressing A for auto-zoom. It may help if you switch on some averaging, so change the box that says "No" to "Mov" then set the number 1 next to it to something like "10". Then send an updated pic if still not working for you.

f4exb, 2022-08-08:
I have a HackRF and could try it myself but I'm pretty sure the issue is with what is mentioned just above. HackRF is known to have a relatively fat noise around DC so the DC block does not block the noise entirely. Moreover depending on the modulation index of the FM I am not sure you can always remove DC without distortion.

darknebular, 2022-08-08:
I have 8 cores, 5Ghz. Can handle without problem until 16-18 Msps.
I put 10 MS/s in order to do the same sample rate than SDR#

The DC offset in SDR# I can´t see this option there, but I can receive a clear voice, but not in SDRAngel.

Yeah, in SDRAngel I need to select the option DC Offset filter, because I can´t hear anything.

Whit the trick of playing with delta frecuency in the demulator and DC Offset option disabled, the problem is the same. The demulator of NFM is very noise and I can´t hear clear the voices.

srcejon, 2022-08-08:
Please post a screenshot showing your new settings, including a nice zoomed in spectrum.

Why is FMd set wider than RF bandwidth?

f4exb, 2022-08-09:
As far as I can see in the screenshot your squelch is open but there is no signal which means that in FM you will just hear loud noise. In FM this is what the squelch is for: shutting down audio when there is no signal. Also you are using the ¨delta" squelch which adjusting is a bit more touchy than plain level squelch so I would advise to de-select the &Delta; button and adjust the level above the level reported in the right corner. In this example say... -50 dB:
[image]

In SDR# you have the I/Q correction checked. It is not possible to do I/Q correction without DC cancelling if things are properly implemented and I believe it is in SDR#. But SDR# also does things without telling you so you don't know if the DC point has been sent outside the passband that you see on its spectrum.

darknebular, 2022-08-09:
> Why is FMd set wider than RF bandwidth?

It´s 8k of bandwidth in the FMd no?
I have 625k of bandwidth. I am using decimate in order to supply the lack of bits in the ADC and have more detailed spectrum, I am wrong with this?

It´s wrong?

Thanks for the replies.

srcejon, 2022-08-09:
> > Why is FMd set wider than RF bandwidth?
> 
> It´s 8k of bandwidth in the FMd no? 

In the picture above, RF bandwidth is 8kHz, but FMd is +-5kHz. I would expect FMd to be at most RF/2, probably less. (So no more than +-4kHz)

If you look at the channel spacing presets, RFBW of 8kHz defaults to FMd of 1.2kHz.

https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodnfm/readme.md

> It´s wrong?

If you post an image showing a close of the signal, it would be easier to tell.

darknebular, 2022-08-09:
It was DC OFFSET

Now I can do a clear hearing:

[image]

darknebular, 2022-08-10:
Is it a design flaw that the DC Offset exists?
Can we use decimation to get more bits of resolution instead of the 8 bits of the ADC?
It seems that removing the DC Offset also removed some of the NFM information.
The solution of playing with the delta frequency is not elegant but it works great, is it the recommended norm in the hackrf or in all SDR radios?

Best regards.

srcejon, 2022-08-10:
> Is it a design flaw that the DC Offset exists?

It's a common problem in direct conversion receivers.

> Can we use decimation to get more bits of resolution instead of the 8 bits of the ADC?

Yes, but that's unrelated to why you see a DC offset.

> It seems that removing the DC Offset also removed some of the NFM information. 

That's because it's removed by filtering. It you have your signal at the same frequency (0Hz), then it will get filtered as well.

>The solution of playing with the delta frequency is not elegant but it works great, 
>is it the recommended norm in the hackrf or in all SDR radios?

It's normal for many SDRs. Some SDRs will hide the problem by adjusting the frequencies by itself.

darknebular, 2022-08-10:
Perfect, thanks for all.
Have a nice day.

closed 2022-08-10 by darknebular