f4exb/sdrangel #1377: Remote TCP Input - FFT does not fix offset on axis X

moffa90, 2022-08-05:
Hey nice work on RemoteTCP Sink. I was using it and i get notice that the FFT graph on axis X doesn't fix accordding to the offset i setup on the plugin parameters, always is centered on the device center frequency e.g: 

- Center frequency device: 482.500 kHz
- Remote TCP sink offset: + 150.000 Hz

With these configuration the FFT graph is centering on 482.500, i guess it should be on 482.650 kHz right? or is this the behavior intended.  I am attaching an image of the escenario to support my comment in case i couldn't explain properly 🐱 

Thank you... 👍 

[image]
?

srcejon, 2022-08-05:
In my mind at least, the frequency offset should only be used if decimation is enabled (and you have that set to 1 in the pic)

moffa90, 2022-08-05:
In fact the signal is being decimated, maybe i missed some details on my last post than can help. The original signal is 2Ms/s, i am down converting it to 48khz, the signal you see on the image is being generated on 482.650 kHz but the FFT graph is centering it on 482.500 where the SDR center freq is configured, so that is the reason i guess the offset should be considered. If we see the FFT it says to me that the signal is at 482.500kHz when is not.

srcejon, 2022-08-08:
Yep, didn't look closely enough at your screenshot. Will fix.

moffa90, 2022-08-08:
Thank you... :+1:

f4exb, 2022-08-14:
Implemented in v7.6.2

closed 2022-08-14 by f4exb