f4exb/sdrangel #1378: Warning/error during startup

loewal, 2022-08-06:
Can someone point me in the direction of avoiding this message? SDRA works well by the way.

2022-08-06 06:52:21.688 (W) PluginManager::loadPluginsDir: Cannot load library /usr/local/lib/sdrangel/plugins/libdemodvorsc.dylib: (dlopen(/usr/local/lib/sdrangel/plugins/libdemodvorsc.dylib, 0x0085): Symbol not found: (__ZN11SWGSDRangel16SWGChannelReport19getVorDemodScReportEv)
  Referenced from: '/usr/local/lib/sdrangel/plugins/libdemodvorsc.dylib'
  Expected in: '/usr/local/lib/sdrangel/libswagger.dylib')

srcejon, 2022-08-06:
Delete /usr/local/lib/sdrangel/plugins/libdemodvorsc.dylib

As it's obsolete (from an old build)

loewal, 2022-08-06:
Thanks Jon,

That did it!

closed 2022-08-06 by srcejon