f4exb/sdrangel #1384: Unable to place "Presets" inside the workspace, like in the older versions.

Daytona3522, 2022-08-09:
Can anyone give me a hint how I can get the 'Presets" back at the same place inside the workspace like the version 6.19.0 screenshot shows below?
I cannot find a way to do this in version 7.6.1. 

[image]

In the new version I can only open it by pressing the "Device Set Presets" button, which is a pain in the ***, and not very useful. 
It would be great if I can place the "Preset" open inside the workspace like shown above.

[image]

Thanks!

f4exb, 2022-08-09:
That's having the preset window permanently on screen that is a pain in the ***... Do you really need to sit permanently once you're done choosing the preset? That's a complete waste of space and this is why it is now implemented as a dialog that makes much more sense. You are the first person complaining so far since the release of v7 several months ago so I can tell you this is not going to change,

Daytona3522, 2022-08-09:
As you may have noticed, it was a question and not a complaint and I trusted you know the difference. I was just asking if anyone knows how to get it back where i like it to be if possible and, not a change request.

f4exb, 2022-08-09:
Well if this was a question then the answer is "no this is not possible"

Daytona3522, 2022-08-09:
All I wanted to know, thanks.

closed 2022-08-13 by f4exb