f4exb/sdrangel #1389: Lime USB TX

srcejon, 2022-08-19:
Hi,

While experimenting with your latency patch, I've noticed a few unrelated things with Lime USB TX.

- Without knowing enough about how the Lime works, I was trying to transmit below 30MHz just by setting the centre frequency to sub <30MHz without using the NCO. This resulted in the transmitted signal being at 30MHz despite the 29MHz setting in SDRangel. After a bit of debugging, I noticed in the log:

    SetFrequencySXT(29 MHz) - required VCO frequency is out of range [3800-7714] MHz

And

    Tx Calibration: MCU error 3 (SXR tune failed)
    2022-08-19 13:01:40.412 (D) LimeSDROutput::applySettings: calibration successful on Tx channel 0

Should we restrict the centre frequency dial to a minimum of 30MHz when the NCO option isn't enabled? And when it is enabled, to 30M+NCO freq? So that a user can't set a value that can't be achieved.

Why does applySettings report calibration as successful when there has been an error?

- I see calibration errors if sample rate is set below 2500000. E.g. at SR=2000000 in the log I get 

    Calibrating Tx for 2.5 MHz (requested 2 MHz [out of range])
    Tx Calibration: MCU error 3 (SXR tune failed)
    2022-08-19 12:21:15.599 (D) LimeSDROutput::applySettings: calibration successful on Tx channel 0

Should the calibration b/w be set differently to SR? Or perhaps b/w just set to a minimum of 2.5MHz?

Should we perhaps add a calibration status indicator, next to U O D in the GUI, that indicates calibration success or failure, so GUI only users can more easily see there has been some sort of problem?

Just as another little bit of help for the user, should the LPF dial background be set to red if at a much smaller value than the NCO? (Indicating that the transmitted signal is being filtered)

srcejon, 2022-08-21:
> Why does applySettings report calibration as successful when there has been an error?

It seems the LimeSuite API docs are wrong. Bug added here: https://github.com/myriadrf/LimeSuite/issues/364

Call to LMS_Calibrate in limesdroutput.cpp should check != 0 rather than < 0.

f4exb, 2022-08-27:
So what remains is:
  - Calibration error indicator in the group of indicators at the bottom. Could be letter "C¨
  - Indication that the bandwidth around LO does not overlap with the baseband around LO + NCO. Instead of the dial numbers turning red maybe just the background of "LP" label should turn red.

closed 2022-08-27 by srcejon