f4exb/sdrangel #1392: Unable to install libdemodadsb.so in  /opt/build/sdrangel/lib/sdrangel/plugins/libdemodadsb.so

Mudary, 2022-08-24:
Installed/compiled https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux
root@mudary:/opt/build/sdrangel/lib/sdrangel/plugins# ls -l
total 11084
-rw-r--r-- 1 root root 2971656 Aug 23 15:51 libbeamsteeringcwmod.so
-rw-r--r-- 1 root root       0 Aug 23 15:52 libdemodadsb.so
-rw-r--r-- 1 root root 4556864 Aug 23 15:51 libdoa2.so
-rw-r--r-- 1 root root 3815328 Aug 23 15:51 libinterferometer.so
root@mudary:/opt/build/sdrangel/lib/sdrangel/plugins# 

Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /opt/build/sdrangel/bin/sdrangelbench
-- Up-to-date: /opt/build/sdrangel/bin/sdrangel
-- Up-to-date: /opt/build/sdrangel/bin/sdrangelsrv
-- Up-to-date: /opt/build/sdrangel/share/sdrangel/Readme.md
-- Up-to-date: /opt/build/sdrangel/share/applications/sdrangel.desktop
-- Up-to-date: /opt/build/sdrangel/share/icons/hicolor/scalable/apps/sdrangel_icon.svg
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libfcdlib.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libfcdhid.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libsdrbase.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libhttpserver.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/liblogging.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libqrtplib.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libswagger.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libhackrfdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/liblimesdrdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libxtrxdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libplutosdrdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libperseusdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libsoapysdrdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libusrpdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libmetisdevice.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libsdrbench.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libmodemm17.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/libsdrgui.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/plugins/libbeamsteeringcwmod.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/plugins/libdoa2.so
-- Up-to-date: /opt/build/sdrangel/lib/sdrangel/plugins/libinterferometer.so
-- Installing: /opt/build/sdrangel/lib/sdrangel/plugins/libdemodadsb.so
CMake Error at plugins/channelrx/demodadsb/cmake_install.cmake:55 (file):
  file RPATH_CHANGE could not write new RPATH:

    /opt/build/sdrangel/lib/sdrangel://opt/build/serialdv/lib://opt/build/libsigmf/lib

  to the file:

    /opt/build/sdrangel/lib/sdrangel/plugins/libdemodadsb.so

  The file format is not recognized.
Call Stack (most recent call first):
  plugins/channelrx/cmake_install.cmake:47 (include)
  plugins/cmake_install.cmake:48 (include)
  cmake_install.cmake:130 (include)

make: *** [Makefile:120: install] Error 1
root@mudary:/opt/build/sdrangel/build# 

Using Ubuntu 20.04.1 Jammy 64-bit
I3-4130 CPU
Mesa 4400 HD Graphics
4 GB memory
1 TB hd

srcejon, 2022-08-24:
Delete the file and run make all - I suspect there's an earlier error you've missed

Mudary, 2022-08-24:
In the root (sudo su), /opt/build, rm -rf sdrangel, then installed with:
cd /opt/build
git clone https://github.com/f4exb/sdrangel.git
cd sdrangel
mkdir build
cd build
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DMIRISDR_DIR=//opt/build/libmirisdr \
-DAIRSPY_DIR=//opt/build/libairspy \
-DAIRSPYHF_DIR=//opt/build/libairspyhf \
-DBLADERF_DIR=//opt/build/libbladeRF \
-DHACKRF_DIR=//opt/build/libhackrf \
-DRTLSDR_DIR=//opt/build/librtlsdr \
-DLIMESUITE_DIR=//opt/build/LimeSuite \
-DIIO_DIR=//opt/build/libiio \
-DPERSEUS_DIR=//opt/build/libperseus \
-DXTRX_DIR=//opt/build/xtrx-images \
-DSOAPYSDR_DIR=//opt/build/SoapySDR \
-DUHD_DIR=//opt/build/uhd \
-DAPT_DIR=//opt/build/aptdec \
-DCM256CC_DIR=//opt/build/cm256cc \
-DDSDCC_DIR=//opt/build/dsdcc \
-DSERIALDV_DIR=//opt/build/serialdv \
-DMBE_DIR=//opt/build/mbelib \
-DCODEC2_DIR=//opt/build/codec2 \
-DSGP4_DIR=//opt/build/sgp4 \
-DLIBSIGMF_DIR=//opt/build/libsigmf \
-DDAB_DIR=//opt/build/libdab \
-DCMAKE_INSTALL_PREFIX=/opt/build/sdrangel ..
make -j $(nproc)
make install
ldconfig

and it works.

Mudary, 2022-08-24:
Closing, because it is now working, thanks to srcejon

closed 2022-08-24 by Mudary