f4exb/sdrangel #1397: Lime SDR mini compatibility

iw4blg, 2022-08-29:
Hello, a friend of mine has a Lime SDR mini HW 1.1 and he's willing to use SDR angel with it. As far as I could understand, the latest version V7 does not support mini. Do you know if the earlier version did it and in the case where I can download it?
Many thanks for all the help!
73, Pierluigi IW4BLG

srcejon, 2022-08-29:
Have you tried it? What error do you get? On what OS?

Valstee, 2022-09-01:
Hi, 
I use a LimeSDR mini with version 6.20 of SDRangel on Ubuntu 20.04 and it works perfectly.
The only error I have is when starting SDRangel in the logs, I see an error related to the calibration of the limesdr mini (if I understand correctly). I think I need a source at 10MHz to do the calibration but it works very well! I haven't tried with the new versions of sdrangel yet.
73, Valentin F4JIZ

justindthomas, 2022-09-02:
I just compiled the newest version on my Pi 400 with a LimeSDR Mini - I've used it for years with no trouble. I didn't realize that version 7 had changed so drastically.

The current version at `master` does not work properly with the LimeSDR Mini. It recognizes the device, but the ranges are all wrong. Here's a quick screenshot.

[image]

It won't allow the frequency to go below 4,294,516. The startup log shows that it initializes the ranges like this:

```
2022-09-01 17:09:03.522 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-01 17:09:03.522 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 30720000.000000
2022-09-01 17:09:03.523 (D) LimeSDRInput::getLPRange: min: 1400100.000000 max: 130000000.000000
2022-09-01 17:09:03.523 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-01 17:09:03.523 (D) LimeSDRInputGUI::updateFrequencyLimits: delta: 0 min: 30000 max: 3500000
2022-09-01 17:09:03.524 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 30720000.000000
2022-09-01 17:09:03.526 (D) LimeSDRInput::applySettings: clock gen frequency:  6.144e+07
```
I'm not sure what those look like normally, but they seem like they're an order of magnitude off to me.

justindthomas, 2022-09-02:
Ah, I see. I had to add it as a `MIMO` device instead of an `Rx` device. That seems to have put me on the right path.
[image]

srcejon, 2022-09-02:
> ```
> 2022-09-01 17:09:03.522 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
> 2022-09-01 17:09:03.522 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 30720000.000000 [...]

The values for Lime USB are:

LimeSDRInput::getLORange: min: 100000.000000 max: 3800000000.000000
LimeSDRInput::getSRRange: min: 100000.000000 max: 61440000.000000
LimeSDRInput::getLPRange: min: 1400100.000000 max: 130000000.000000
LimeSDRInput::getLORange: min: 100000.000000 max: 3800000000.000000
LimeSDRInputGUI::updateFrequencyLimits: delta: 0 min: 30000 max: 3800000
LimeSDRInput::getSRRange: min: 100000.000000 max: 61440000.000000

However, the Lime Mini has a reduced frequency range of 10MHz to 3.5GHz, so the values above look OK to me.

> Ah, I see. I had to add it as a MIMO device instead of an Rx device

The Rx device should be able to work.

srcejon, 2022-09-02:
I notice in your first screenshot, you have the NCO enabled and external clock enabled - can I suggest you switch those off to see what happens?

Valstee, 2022-09-02:
I just tested with Ubuntu 22.04 and Sdrangel 7.6.2, I have no problem everything works perfectly. I don't have the calibration error message I mentioned before and I use it as an RX device (not MIMO).

Valstee, 2022-09-02:
in fact if I have indeed a message saying :
```
Rx calibration: MCU error 6 (invalid Rx path)
2022-09-02 15:35:07.439 (C) LimeSDRInput::applySettings: calibration failed on Rx channel 0
```
but everything seems to work perfectly. I already had this message on Sdrangel versions 6 (and previous)

srcejon, 2022-09-02:
> in fact if I have indeed a message saying :
> 
> ```
> Rx calibration: MCU error 6 (invalid Rx path)
> 2022-09-02 15:35:07.439 (C) LimeSDRInput::applySettings: calibration failed on Rx channel 0
> ```
> 
> but everything seems to work perfectly. I already had this message on Sdrangel versions 6 (and previous)

In 7.6.3 this will probably result in a red status light in the GUI.

On the Lime USB, I get this same error if antenna is set to No (None), which is the default value. What antenna do you have selected?

What is odd, is that in the screenshot above, the antenna combo selection is blank. Can you check if the list has Lo and Hi in it? One of those should be selected.

Valstee, 2022-09-02:
Thanks for the information about the choice of the antenna. I had just run sdrangel and limesdr to test in relation to this issue. Everything works for me now ! Thanks 

```
Rx calibration finished
2022-09-02 16:26:37.852 (D) LimeSDRInput::applySettings: calibration successful on Rx channel 0
```

justindthomas, 2022-09-02:
@srcejon for some reason, the NCO and External Clock selections are lit by default when I add the Rx device. I can turn off the NCO by clicking it, but clicking the External Clock button just opens a dialog box.

[image]

[image]

Here is the [Startup Log](https://gist.github.com/justindthomas/11dd9a16b9a695fb0f2e0169144c291b).

srcejon, 2022-09-02:
In the pop up box, set the frequency to 10,000,000 rather than 300,000,000.

Also check the transverter X box and set the delta frequency to 0.

It's a bit odd that the external clock is checked in the Device GUI, but the Tick button isn't checked in the popup dialog. Definitely something amiss.

justindthomas, 2022-09-02:
Strange. I can change the Transverter frequency to 0 (it defaults to +4,284,516,145) but when I change the External Clock frequency to 10,000,000, the Transverter frequency reverts to that +4,284,516,145. I can't seem to set both of those to the desired values (0 and 10,000,000) at the same time.

justindthomas, 2022-09-02:
And the NCO button keeps turning on by itself. If I turn it off and then go to change the External Clock to 10,000,000, it gets turned back on and the clock reset to 300,000,000 as soon as I close it.

```
2022-09-02 10:32:24.245 (D) LimeSDRInputGUI::on_extClock_clicked: 300000000 Hz off
2022-09-02 10:32:24.386 (D) LimeSDRInputGUI::updateHardware
2022-09-02 10:32:24.386 (D) LimeSDRInput::handleMessage: MsgConfigureLimeSDR
2022-09-02 10:32:24.388 (D) LimeSDRInput::applySettings: clock gen frequency:  4.612e+09
Error
2022-09-02 10:32:24.417 (D) LimeSDRInput::applySettings: clock set to internal (Ext: 300000000 Hz)
2022-09-02 10:32:24.420 (D) LimeSDRInput::applySettings: clock gen frequency after:  4.612e+09
2022-09-02 10:32:24.420 (D) LimeSDRInput::suspendRxBuddies (0)
2022-09-02 10:32:24.420 (D) LimeSDRInput::suspendTxBuddies (0)
Rx calibration: MCU error 4 (SXT tune failed)
2022-09-02 10:32:24.486 (C) LimeSDRInput::applySettings: calibration failed on Rx channel 0
2022-09-02 10:32:24.487 (D) LimeSDRInput::resumeTxBuddies (0)
2022-09-02 10:32:24.487 (D) LimeSDRInput::resumeRxBuddies (0)
2022-09-02 10:32:24.487 (D) LimeSDRInput::applySettings: center freq:  281473699822840  Hz  m_transverterMode:  true  m_transverterDeltaFrequency:  4284516145  deviceCenterFrequency:  281469415306695  device stream sample rate:  0 S/s  sample rate with soft decimation:  0 S/s  ADC sample rate with hard decimation:  0 S/s  m_log2HardDecim:  0  m_log2SoftDecim:  0  m_iqOrder:  false  m_gain:  4284516303  m_lpfBW:  0  m_lpfFIRBW:  -2,147,483,648  m_lpfFIREnable:  true  m_ncoEnable:  true  m_ncoFrequency:  0  m_antennaPath:  -1276888384  m_extClock:  false  m_extClockFreq:  300,000,000  m_gpioDir:  0  m_gpioPins:  0  force:  false  forceNCOFrequency:  false  doCalibration:  true  doLPCalibration:  false
2022-09-02 10:32:24.491 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-02 10:32:24.492 (D) LimeSDRInputGUI::updateFrequencyLimits: delta: 4284516 min: 4294516 max: 7784516
2022-09-02 10:32:24.492 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 30720000.000000
2022-09-02 10:32:24.494 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-02 10:32:24.494 (D) LimeSDRInputGUI::updateFrequencyLimits: delta: 4284516 min: 4294516 max: 7784516
2022-09-02 10:32:24.599 (D) LimeSDRInputGUI::updateHardware
2022-09-02 10:32:24.599 (D) LimeSDRInput::handleMessage: MsgConfigureLimeSDR
2022-09-02 10:32:24.601 (D) LimeSDRInput::applySettings: clock gen frequency:  4.612e+09
Using external reference clock requires hardware modification (R59/R62)
2022-09-02 10:32:24.601 (D) LimeSDRInput::applySettings: clock set to external (Ext: -1292363136 Hz)
2022-09-02 10:32:24.603 (D) LimeSDRInput::applySettings: clock gen frequency after:  4.612e+09
2022-09-02 10:32:24.603 (D) LimeSDRInput::suspendRxBuddies (0)
2022-09-02 10:32:24.603 (D) LimeSDRInput::suspendTxBuddies (0)
Rx calibration: MCU error 4 (SXT tune failed)
2022-09-02 10:32:24.685 (C) LimeSDRInput::applySettings: calibration failed on Rx channel 0
2022-09-02 10:32:24.685 (D) LimeSDRInput::resumeTxBuddies (0)
2022-09-02 10:32:24.685 (D) LimeSDRInput::resumeRxBuddies (0)
2022-09-02 10:32:24.686 (D) LimeSDRInput::applySettings: center freq:  281473699822840  Hz  m_transverterMode:  true  m_transverterDeltaFrequency:  4284516145  deviceCenterFrequency:  281469415306695  device stream sample rate:  0 S/s  sample rate with soft decimation:  0 S/s  ADC sample rate with hard decimation:  0 S/s  m_log2HardDecim:  0  m_log2SoftDecim:  0  m_iqOrder:  false  m_gain:  4284516303  m_lpfBW:  0  m_lpfFIRBW:  -2,147,483,648  m_lpfFIREnable:  true  m_ncoEnable:  true  m_ncoFrequency:  0  m_antennaPath:  -1276888384  m_extClock:  true  m_extClockFreq:  3,002,604,160  m_gpioDir:  0  m_gpioPins:  0  force:  false  forceNCOFrequency:  false  doCalibration:  true  doLPCalibration:  false
2022-09-02 10:32:24.688 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-02 10:32:24.689 (D) LimeSDRInputGUI::updateFrequencyLimits: delta: 4284516 min: 4294516 max: 7784516
2022-09-02 10:32:24.689 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 30720000.000000
2022-09-02 10:32:24.689 (D) LimeSDRInput::getLORange: min: 10000000.000000 max: 3500000000.000000
2022-09-02 10:32:24.690 (D) LimeSDRInputGUI::updateFrequencyLimits: delta: 4284516 min: 4294516 max: 7784516
2022-09-02 10:32:24.794 (D) LimeSDRInputGUI::updateHardware
2022-09-02 10:32:24.794 (D) LimeSDRInput::handleMessage: MsgConfigureLimeSDR
2022-09-02 10:32:24.796 (D) LimeSDRInput::applySettings: clock gen frequency:  4.612e+09
2022-09-02 10:32:24.798 (D) LimeSDRInput::applySettings: clock gen frequency after:  4.612e+09
2022-09-02 10:32:24.798 (D) LimeSDRInput::applySettings: center freq:  281473699822840  Hz  m_transverterMode:  true  m_transverterDeltaFrequency:  4284516145  deviceCenterFrequency:  281469415306695  device stream sample rate:  0 S/s  sample rate with soft decimation:  0 S/s  ADC sample rate with hard decimation:  0 S/s  m_log2HardDecim:  0  m_log2SoftDecim:  0  m_iqOrder:  false  m_gain:  4284516303  m_lpfBW:  0  m_lpfFIRBW:  -2,147,483,648  m_lpfFIREnable:  true  m_ncoEnable:  true  m_ncoFrequency:  0  m_antennaPath:  -1276888384  m_extClock:  true  m_extClockFreq:  3,002,604,160  m_gpioDir:  0  m_gpioPins:  0  force:  false  forceNCOFrequency:  false  doCalibration:  false  doLPCalibration:  false
```

srcejon, 2022-09-02:
@Valstee are you on Arm or x64?

justindthomas, 2022-09-02:
Fwiw, the MIMO setting seems to work well, and performance is really improved quite a bit since 6.2.x (less flickering of the waterfall and spectrum graphs).

closed 2022-11-08 by github-actions[bot]