f4exb/sdrangel #1406: SSTV modulator/demodulator

AsciiWolf, 2022-09-12:
It would be great if SDRangel had a slow-scan television modulator and demodulator. Especially on Linux where there aren't many applications for SSTV.

f4exb, 2022-09-14:
What about QSSTV: https://github.com/xdsopl/qsstv I saw your name there 😄

AsciiWolf, 2022-09-14:
That GitHub repo is sadly just an unofficial fork. The official development page [was here](http://users.telenet.be/on4qz/), but it looks like that the development has died and the page is not available anymore.

f4exb, 2022-09-14:
Maybe the fork has still valuable code I did not look into details but it is Qt based so it could be a good start. There are many SSTV standards and I do not have a lot of samples at hand so it is interesting to rely on proven code.

f4exb, 2022-09-21:
I'll try to build and play with QSSTV to see how it looks like. May be useful to get some test signals also.

srcejon, 2022-09-21:
I've just stuck a PD120 SSTV capture from the ISS here: http://www.sdrangel.org/iq/sstv.zip

Bit of a drop-out in signal strength in the middile, but strong at the start.

sarvet, 2023-04-14:
Two resources I've found: 
https://github.com/ON4QZ/QSSTV
https://www.qsl.net/on4qz/
According to [this Post on Reddit](https://www.reddit.com/r/amateurradio/comments/ybzolm/anyone_know_what_happened_to_qsstv/):  "The provider Telenet has unilaterally stopped the Web service. Not even giving me a chance to put a redirect. […]"

AsciiWolf, 2023-04-14:
Yep, I forgot to mention that the QSSTV was moved to https://www.qsl.net/on4qz/, so there is no need to use a fork or a Debian archive anymore.

pchmielewski1, 2025-03-09:
3 years later, still no sdrangel support for sstv, I have to do a workaround to do it another way, I know free software I can't require, I don't recommend, often restarts. unstable software.