f4exb/sdrangel #1407: Error at compiling on M1 Mac

mrponzo, 2022-09-12:
hello! 
Getting this error when trying to compile to be able to use the ADALM-PLUTO.
None of the binaries seems to give me the option... 
Anyways... the error is this:

-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.32.1 (Apple Git-133)")
-- SDRangel Version: 7.6.4
-- Build type set to: Release
-- Build for macOS target: local version
-- RPATH support: ON
-- Compiling for 24 bit Rx DSP chain
-- Looking for __x86_64__
-- Looking for __x86_64__ - not found
-- Looking for __i386__
-- Looking for __i386__ - not found
-- Looking for __arm__
-- Looking for __arm__ - not found
-- Looking for __aarch64__
-- Looking for __aarch64__ - found
-- Target architecture: ARM64-native
-- Performing Test FLAG_SUPPORTED
-- Performing Test FLAG_SUPPORTED - Failed
CMake Error at cmake/Modules/DetectArchitecture.cmake:192 (message):
  Flag '-march=native' rejected by compiler.  Please adjust ARCH_OPT option.
Call Stack (most recent call first):
  cmake/Modules/CompilerOptions.cmake:3 (include)
  CMakeLists.txt:396 (include)

-- Configuring incomplete, errors occurred!
See also "/opt/build/sdrangel/build/CMakeFiles/CMakeOutput.log".
See also "/opt/build/sdrangel/build/CMakeFiles/CMakeError.log".

Thanks in advance!

srcejon, 2022-09-13:
You need clang 15 for -march=native to be supported:

https://www.phoronix.com/news/LLVM-Clang-march-native-M1

Failing that, try something like ARCH_OPT=   or ARCH_OPT=arm64

ReDetection, 2022-09-27:
I can confirm commenting out `ARCH_OPT "native"` in CMakeLists.txt still does help. Reference #1121

closed 2022-12-03 by github-actions[bot]