f4exb/sdrangel #1409: APRS source does not see MIMO sourced packet demodulator

audriusb, 2022-09-15:
If i open up my LimeSDR with MIMO and add a pkt demodulator, this does not appear in APRS sources list.
If opened as single RX device: everything is fine. 

(Though i get segfaults, time to time on simple RX with limesdr, but i blame ubuntu 22.04 for that, since I compiled sdrangel myself and this does not have any issues with rtlsdr). 

Version: 7.6.3

srcejon, 2022-09-15:
When APRS::scanAvailableChannels() iterates through deviceSets, it only looks for DSPDeviceSourceEngine, not DSPDeviceMIMOEngine.

srcejon, 2022-09-15:
The above patch should fix the Packet Demod not appearing in the sources list.

I can't receive APRS right now, so do not know if it will work completely - so please let me know if you still have a problem.

audriusb, 2022-09-16:
Works like a charm. Thanks for the speed. 
[image]

closed 2022-09-16 by audriusb