f4exb/sdrangel #1416: Upgrade Qt version for Ubuntu build

f4exb, 2022-09-17:
It is becoming an increasing pain to maintain Qt 5.12 compatibility because Ubuntu builds are based on 20.04. Possible solutions:

* In Ubuntu 20.04 image it may be possible to use Qt 5.15: https://www.appveyor.com/docs/linux-images-software/#qt
* move CI to Github actions where an Ubuntu 22.04 with Qt 5.15 is available

f4exb, 2022-09-17:
Using Qt5.15 on the 20.04 image is a no-no the image does not seem to contain any Qt pre-installed software. As it runs on premises it appears to be a "minimal" Ubuntu image as mentioned in the image tag. However some other options could be investiagated:

* The image used is tagged as appveyor-byoc-linux:latest so any image with this tag would be picked up by the Appveyor agent. Now the issue is to build this image correctly based on a more recent distro (Ubuntu 22.04)
* As it already runs on premises one could also go for full Appveyor server hosting. See: https://www.appveyor.com/docs/server/ Then it seems the build images can be fully customized

Note: this is the docker ps display of a running build:
```
CONTAINER ID                                                       IMAGE                 COMMAND                                                 CREATED          STATUS                   PORTS     NAMES                                  
90f76d5b88c746f830aff2671f09189d50e1c7b6a70942a54ac00eee36b4a463   appveyor-byoc-linux   "/bin/bash /scripts/entrypoint.sh"                      11 minutes ago   Up 11 minutes                      admiring_poincare 
```

srcejon, 2022-09-17:
CI is one consideration, but what might users want? If we require Qt 5.15, are there any major distros that don't have 5.15 on their most recent LTS equivalent release?

Quick look at the versions in the VMs I have:

Ubuntu 20.04 - 5.12.8
Ubuntu 21.10 - 5.15.2
Ubuntu 22.04 - 5.15.3
Debian 10 - 5.11
Debian 11 - 5.15.2
Centos 7 - 4
Centos 8 - 5.15.2
Fedora 36 - 5.15.3
Windows - 5.15.2
Mac - 5.15.2

So probably OK, unless we want to support older distros.

f4exb, 2022-09-17:
Indeed 5.15 seems to be in every modern distribution. I think we should drop 5.12 compatibility anyway as it is holding the code too much back.

f4exb, 2022-09-18:
Now the CI builds an Ubuntu 22.04 .deb

f4exb, 2022-09-18:
Hack to create the Ubuntu 22.04 image used for builds: https://github.com/f4exb/appveyor-build-images/tree/ubuntu2204_hack
Run the `linuxbuild.sh` script
Missing is tagging of the image: `docker tag <image id> appveyor/build-image:minimal-ubuntu-22.04`
Then to make it the default image used by appveyor agent: `docker tag <image id> appveyor-byoc-linux:latest`

f4exb, 2022-09-18:
Documentation updated