f4exb/sdrangel #1417: MacOs compilation error

loewal, 2022-09-17:
[ 31%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/util/iot/visa.cpp.o
/Users/Loewal/Downloads/sdrangel/sdrbase/util/iot/visa.cpp:266:84: error: cannot pass object of non-trivial type 'const QString' through variadic function; call will abort at runtime [-Wnon-pod-varargs]
                QString commands = QString::asprintf(control->m_setState.toUtf8(), state);
                                                                                   ^
1 error generated.
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/util/iot/visa.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [all] Error 2

What could be the reason.....?

srcejon, 2022-09-17:
Try: 

QString commands = QString::asprintf(control->m_setState.toUtf8(), state.toUtf8());

loewal, 2022-09-17:
Sorry Jon,
Where should I issue this setting/command?

loewal, 2022-09-17:
Changed the error-line...
[ 21%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/util/iot/visa.cpp.o
/Users/Loewal/Downloads/sdrangel/sdrbase/util/iot/visa.cpp:266:70: error: cannot pass object of non-trivial type 'QByteArray' through variadic function; call will abort at runtime [-Wnon-pod-varargs]
                QString commands = QString::asprintf(control->m_setState.toUtf8(), state.toUtf8());
                                                                                   ^
1 error generated.
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/util/iot/visa.cpp.o] Error 1
make[1]: *** [sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [all] Error 2

srcejon, 2022-09-17:
Sorry, that should be:

                QString commands = QString::asprintf(control->m_setState.toUtf8(), state.toUtf8().data());

loewal, 2022-09-17:
Thanks Jon,
That was it indeed!

srcejon, 2022-09-17:
Thanks for trying the fixes.

loewal, 2022-09-17:
Happy to early Jon,

/Users/Loewal/Downloads/sdrangel/plugins/feature/remotecontrol/remotecontrolgui.cpp:1008:61: error: cannot pass object of non-trivial type 'QByteArray' through variadic function; call will abort at runtime [-Wnon-pod-varargs]
        formattedValue = QString::asprintf(format.toUtf8(), value.toString().toUtf8());
                                                            ^
1 error generated.
make[2]: *** [plugins/feature/remotecontrol/CMakeFiles/featureremotecontrol.dir/remotecontrolgui.cpp.o] Error 1
make[1]: *** [plugins/feature/remotecontrol/CMakeFiles/featureremotecontrol.dir/all] Error 2
make: *** [all] Error 2

Solution:

1008         formattedValue = QString::asprintf(format.toUtf8(), value.toString().toUtf8().data());

ok?

Compiled well after above fix at line 1008

srcejon, 2022-09-18:
> Solution:
> 
> 1008 formattedValue = QString::asprintf(format.toUtf8(), value.toString().toUtf8().data());
> 
> ok?

Yep, thanks. I'll add a patch for that too.

closed 2022-09-27 by srcejon