f4exb/sdrangel #1423: problem with 3 D Map

choupette-change, 2022-09-19:
Hello, I just installed sdrangel_7.7.0-1_amd64.deb with Ubuntu 22.04. I tested ADSB and AIS demodulators, with the 3D map (map feature) . If I change the initial default color of the 3D trace, the new color is not displayed??

srcejon, 2022-09-19:
Is your terrain to set to Ellipsoid? If not, the path will probably be underground. Screenshots would help.

choupette-change, 2022-09-19:
Yes my terrain is set to Ellipsoid. This problem is not with windows version. If you change the initial color of this 3 D map feature, the new color is not display (3D track, 3D point, ...) on Ubuntu version.

srcejon, 2022-09-21:
The problem appears to be that the Alpha channel value in the colour picker dialog isn't being set, and is defaulting to 0, which means it is transparent. Make sure you set it to 255:

[image]

choupette-change, 2022-09-21:
It's ok with the color problem but I still have the problem with the 3d map disappearing after a very variable time of operation (a few minutes to several hours..) with the latest version sdrangel_7.7.0-1_amd64.deb.
[image]

choupette-change, 2022-09-22:
Moreover, I don't have this problem with other ADSB, APT, APRS demodulators....

srcejon, 2022-09-22:
Have a look at memory usage - some of the ship models are pretty big.

choupette-change, 2022-09-22:
I just did another try, before the 3D image disappeared, I was able to grab these lines from the log. Previously I had reduced the size of the icons: 3d min pixels= 10. if that helps..
2022-09-22 18:05:13.250 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 16384 samples
2022-09-22 18:05:13.250 (D) NFMDemodSink::processOneSample: 0/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
2022-09-22 18:05:13.594 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 12288 samples
2022-09-22 18:05:13.594 (D) NFMDemodSink::processOneSample: 4096/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
2022-09-22 18:05:13.937 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 12288 samples
2022-09-22 18:05:13.938 (D) NFMDemodSink::processOneSample: 4096/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
[14242:14303:0922/180514.155317:ERROR:shared_image_manager.cc(143)] SharedImageManager::ProduceGLTexture: Trying to produce a representation from a non-existent mailbox. 80:1D:9D:CA:81:92:89:FD:7F:01:6C:C8:17:B5:5E:67
[14242:14303:0922/180514.162964:ERROR:gles2_cmd_decoder.cc(19028)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoCreateAndTexStorage2DSharedImageINTERNAL: invalid mailbox name
[14242:14303:0922/180514.163000:ERROR:gles2_cmd_decoder.cc(19050)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoBeginSharedImageAccessCHROMIUM: bound texture is not a shared image
[14242:14303:0922/180514.163024:ERROR:gles2_cmd_decoder.cc(19079)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoEndSharedImageAccessCHROMIUM: bound texture is not a shared image
[14242:14303:0922/180514.163042:ERROR:gles2_cmd_decoder.cc(19050)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoBeginSharedImageAccessCHROMIUM: bound texture is not a shared image
[14242:14303:0922/180514.163211:ERROR:gles2_cmd_decoder.cc(10695)] [.DisplayCompositor]RENDER WARNING: texture bound to texture unit 0 is not renderable. It might be non-power-of-2 or have incompatible texture filtering (maybe)?
[14242:14303:0922/180514.163279:ERROR:gles2_cmd_decoder.cc(19079)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoEndSharedImageAccessCHROMIUM: bound texture is not a shared image
[14242:14303:0922/180514.163371:ERROR:shared_image_manager.cc(143)] SharedImageManager::ProduceGLTexture: Trying to produce a representation from a non-existent mailbox. 99:12:A7:CA:09:3E:3B:6A:20:B5:8C:29:B3:44:9C:67
[14242:14303:0922/180514.163405:ERROR:gles2_cmd_decoder.cc(19028)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoCreateAndTexStorage2DSharedImageINTERNAL: invalid mailbox name
[14242:14303:0922/180514.163428:ERROR:gles2_cmd_decoder.cc(19050)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoBeginSharedImageAccessCHROMIUM: bound texture is not a shared image
[14242:14303:0922/180514.163465:ERROR:gles2_cmd_decoder.cc(10695)] [.DisplayCompositor]RENDER WARNING: texture bound to texture unit 0 is not renderable. It might be non-power-of-2 or have incompatible texture filtering (maybe)?
[14242:14303:0922/180514.163496:ERROR:gles2_cmd_decoder.cc(19079)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : DoEndSharedImageAccessCHROMIUM: bound texture is not a shared image
2022-09-22 18:05:14.274 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 8192 samples
2022-09-22 18:05:14.274 (D) NFMDemodSink::processOneSample: 8192/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
2022-09-22 18:06:25.953 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 4096 samples
2022-09-22 18:06:25.953 (D) NFMDemodSink::processOneSample: 12288/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
2022-09-22 18:06:26.297 (C) AudioFifo::write: (NFMDemod [0:2]) overflow 16384 samples
2022-09-22 18:06:26.297 (D) NFMDemodSink::processOneSample: 0/16384 audio samples written m_audioSampleRate: 48000 m_channelSampleRate: 64000
2022-09-22 18:06:29.536 (D) RTLSDRInput::handleMessage: MsgStartStop:  stop
2022-09-22 18:06:29.539 (D) DSPDeviceSourceEngine::stopAcquistion
2022-09-22 18:06:29.539 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-09-22 18:06:29.539 (D) DSPDeviceSourceEngine::gotoIdle
2022-09-22 18:06:29.609 (D) GLSpectrumGUI::handleInputMessages: message: SpectrumVis::MsgStartStop
2022-09-22 18:06:29.609 (D) AISDemod::stop
2022-09-22 18:06:29.610 (D) AISDemod::stop
2022-09-22 18:06:29.610 (D) NFMDemod::stop
2022-09-22 18:06:29.612 (D) AudioDeviceManager::removeAudioSink: 0x55eb31d18f20
2022-09-22 18:06:29.615 (D) ChannelAnalyzer::stop
2022-09-22 18:06:29.620 (D) DC offset:0.000000,0.000000

[image]

closed 2022-09-27 by srcejon