f4exb/sdrangel #1424: Airspy HF

loewal, 2022-09-19:
Device is recognised, but nothing happens.
Warning/error: samplerate must be >= 2048000

But there is no way to set this.....

f4exb, 2022-09-19:
Easy answer: the demod you are using with the Airspy HF+ requires a bandwidth of 2048 kHz or more thus a sample rate of at least 2048000 S/s as the message warns you. However the maximum rate of an Airspy HF+ is 768000 S/s so your setup cannot work. A simple RTL-SDR dongle would do though.

loewal, 2022-09-19:
So what you say, is that a different demod should work?
Which demod could that be? e.g. USB?
I was running DAB, so that couldn't work....

loewal, 2022-09-19:
Tried SSB: no error/warning, but no signal whatsoever.
Tried several settings, also lower samprates.
In SdrPlusPlus, CubicSDr anfd GQRX all works fine, but those are not my favorite apps.

loewal, 2022-09-19:
Is it okay by the way that the device is shown as AirspyHF and not AirspyHF+?

> Is this a duplicate of: #1327?

I am sorry...yes it is....

srcejon, 2022-09-20:
Are you building airspyhf library yourself?

The SDRangel Linux compile instructions have:

git reset --hard "1.1.5"

You might try rebuilding it without this command, so you use the latest version, and see if that makes any difference.

loewal, 2022-09-20:
Perhaps worth mentioning:
Under Linux everything is fine.
Only under MacOs nothing happens. See indeed #1327

loewal, 2022-09-20:
And yes, I compiled it with and also without git reset etc.
No change helas.

srcejon, 2022-09-21:
Closed - will continue under 1327

closed 2022-09-21 by srcejon