f4exb/sdrangel #1432: Workspace layout messed up after reopening sdrangel

audriusb, 2022-09-21:
Version: 7.7.0 
Fresh release installed on ubuntu 22.04 from released package. Had a compiled version before, same behaviour.

This was also the same with previous versions, since the workspace concept was introduced (I like it).
Once set up my layout, save conf and reopen sdrangel the layout messed up. All the devices, channels, etc appear far down to the right behind the window limits - need to scroll down and right to see them.
Tried saving with "save all" and with Configurations box. Same results.
Removed config file, recreate the layout, save again - same results.

I should probably mention I work with 4 monitors, so this might be an issue form that, since I don't code in C++ nor know Qt, i'm only guessing here.

f4exb, 2022-09-21:
I am not sure about how this can be controlled. Qt does pretty much what it wants...

audriusb, 2022-09-21:
Like i mentioned, I'm not good at this, but it looks like the widgets are moved to a point with `mode(p)`. Then the question could be: are the positions saved to the config? 
Sorry if this is a complete nonsense that I'm talking.

srcejon, 2022-09-21:
Yes, SDRAngel uses QWidget::saveGeometry QWidget::restoreGeometry to save and restore widget positions, and this works on single monitor setups.

There is this Qt bug: https://bugreports.qt.io/browse/QTBUG-70721 that suggests calling restoreGeometry twice might help if related to DPI scaling.

commit 375c0e1c (2022-09-24): Duplicate QWidget::restoreGeometry in some places in an attempt to fix #1432
closed 2022-09-24 by f4exb