f4exb/sdrangel #1435: Feature Request: Switch from Rx to Tx on VOX

CRD716, 2022-09-21:
I use a Hackrf One for ham radio use, and while it works fine for most non-automated use, trying to use something like FT8 using WSJT-X is difficult, as one has to manually disable the receive source and enable a transmit source. I know there are some sdr specific issues (mainly the requirement of a small buffer during the switching time as mentioned in greatscottgadgets/hackrf/issues/1061), but even a basic "turn x device on/off by vox" would improve things.
If there could be a feature or setting where any noise above a certain db from the microphone (virtual or otherwise) could switch the device to transmit, a much wider variety of features could be possible.
I'm willing to help in any way I can, testing or otherwise. If this feature already exists in some way, my apologies in advance.

f4exb, 2022-09-21:
The "simple" PTT has a VOX capability: https://github.com/f4exb/swagger-codegen/blob/sdrangel/modules/swagger-codegen/src/main/resources/qt5cpp/HttpRequest.cpp.mustache#L157 but as it uses its own buffer it is not necessarily strongly coordinated to the modulator. 

I don't know WSJT-X well enough (only used it on Rx side) but it would be much better if WSJT-X could trigger the Tx switchover via SDRangel API. This could be possible if WSJT-X would launch an O/S command a switchover time then it is trivial to write a Python or a shell script (with curl) that calls the SDRangel API.

f4exb, 2022-09-21:
Apparently there is no such thing with WSJT-X but it can talk to a rigctl server and there is a plugin that does that: https://github.com/f4exb/sdrangel/tree/master/plugins/feature/rigctlserver However at the moment it controls just one "device" and is Rx only and I don't know if the rigctl protocol even supports Rx/Tx switchover.

CRD716, 2022-09-22:
https://manpages.ubuntu.com/manpages/trusty/man8/rigctld.8.html RigCTL seems to support some form of rx/tx switch with  

> T, set_ptt 'PTT'
>               Set 'PTT', 0 (RX) or 1 (TX).

CRD716, 2022-09-22:
misclick on the close

f4exb, 2022-09-22:
Since the VOX feature already exist with the SimplePTT plugin but may not be ideal (you may want to try it though) I'd rather change the requirement for RigCtl plugin as discussed above. I am not the author of RigCtl but I can try to do the changes or Jon @srcejon could do it if he wants to.

srcejon, 2022-09-22:
You're more than welcome to :)

There's this issue as well relating to WSJT-X and RigCtl: https://github.com/f4exb/sdrangel/issues/1199

f4exb, 2022-09-22:
Not sure I would implement TS-2000 protocol as suggested in #1199 but in the course of developing the present feature I will try the WSJT-X - SDRangel Rx/Tx end to end so we will see...

CRD716, 2022-09-22:
Agreed, rigctl could have other benefits as well. (Will be trying the SimplePTT as well)

> Since the VOX feature already exist with the SimplePTT plugin but may not be ideal (you may want to try it though) I'd rather change the requirement for RigCtl plugin as discussed above. I am not the author of RigCtl but I can try to do the changes or Jon @srcejon could do it if he wants to.

f4exb, 2022-09-24:
For WSJT-X the setup should be similar to what is done for Cubic SDR: https://www.radiosrs.net/digital_modes.html

N5BRG, 2022-09-24:
I worked on this issue two years ago. At the time I forked Hamlib (GitHub/n5brg) and wrote code to add to hamlib to control SDRAngel via the http interface. The issue I ran into was the WSJT-X code has a rigid turn around time requirement between TX and RX which I could not meet.  I opened an issue at the time looking for some help on SDRAngel and also talked to WSJT-X developers about relaxing the time. There needs to be a faster interface with SDRAngel to make the rigctl changes work at the rate required by WSJT-X.

For WSJT-X to work properly you must do RX/TX switching but also there is a need to adjust frequency on TX or RX. Many other rigctl commands exist as well.

Hopefully the renewed interest will move this along as WSJT-X is a very popular mode of operation and it does need something like rigctl.

Bob N5BRG

PS: I am not a world class coder like Edouard or Jon.

f4exb, 2022-09-24:
One should use [libfaketime](https://github.com/wolfcw/libfaketime). See example here: https://github.com/f4exb/sdrangel-docker/tree/master/wsjtx