f4exb/sdrangel #1456: Feature Request: Add Opera Cake support (HackRF)

CRD716, 2022-09-29:
A new Hackrf firmware has launched (2022.09.1) which includes support for the opera cake antenna switcher. It would be nice to have a way to use this functionality from within SDRAngel.
Also, in external/CMakeLists.txt hackrf is set to a 2018 build, is this on purpose?
greatscottgadgets/hackrf/releases

CRD716, 2022-09-29:
Also, while the build instructions should work, the git repository has changed from mossman/hackrf for greatscottgadgets/hackrf

CRD716, 2022-09-29:
After some testing, the latest version of both the firmware and lib works with sdrangel. I see no reason not to change the build to the latest version.

f4exb, 2022-09-30:
Upgrading libhackrf version used is a different story but for what concerns Opera Cake I haven't seen any support in libhackrf documentation it seems everything is done from the command line. In this case I do not see the point of having it as something specific the "commnds" functionality can do this for you: https://github.com/f4exb/sdrangel/tree/master/sdrgui#23-commands 

Even if there was support in libhackrf this would have to be a feature plugin like the LimeRFE feature and not something in the HackRF plugins themselves. Not anyone has an Opera Cake add-on card and this would be an annoyance.

Moreover as far as I understood the Opera Cake has an automatic mode that can already do a lot of stuff for you without passing commands like switching on crossing frequency boundaries or switching Rx/Tx.

CRD716, 2022-09-30:
Fair enough, guess this should be switched to updating the libhackrf used for performance and stability purposes.

commit 1c47ebbd (2022-10-06): Upgrade to latest libhackrf and use new Github URL. Fixes #1456
closed 2022-10-06 by f4exb