f4exb/sdrangel #1460: Error building aptdec build

choupette-change, 2022-10-01:
Hello,
tutu@tutu-Aspire-E5-573:/opt/build/aptdec/build$ cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/aptdec ..
-- The C compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.37") 
-- Found LibSndFile: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Configuring done
CMake Error at CMakeLists.txt:28 (add_executable):
  Cannot find source file:

    src/argparse/argparse.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:28 (add_executable):
  No SOURCES given to target: aptdec

CMake Generate step failed.  Build files cannot be regenerated correctly.
tutu@tutu-Aspire-E5-573:/opt/build/aptdec/build$

srcejon, 2022-10-01:
Because you have all of its dependencies, it's trying to build the .exe, not just the .so. In that case, you also need to checkout its submodules:

cd /opt/build/aptdec
git submodule update --init --recursive

Then run cmake.

choupette-change, 2022-10-01:
Thanks for the information. Another detail:
As a general prerequisite you should have the following packages installed:

sudo apt-get update && sudo apt-get -y install \
git cmake g++ pkg-config autoconf automake libtool libfftw3-dev libusb-1.0-0-dev libusb-dev libhidapi-dev libopengl-dev \
qtbase5-dev qtchooser libqt5multimedia5-plugins qtmultimedia5-dev libqt5websockets5-dev \
qttools5-dev qttools5-dev-tools libqt5opengl5-dev qtbase5-dev libqt5quick5 libqt5charts5-dev \
**qml-module-qtlocation  qml-module-qtlocation** qml-module-qtpositioning qml-module-qtquick-window2 \
qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qtquick-controls2 qml-module-qtquick-layouts \
libqt5serialport5-dev qtdeclarative5-dev qtpositioning5-dev qtlocation5-dev libqt5texttospeech5-dev \
qtwebengine5-dev qtbase5-private-dev \
libfaad-dev zlib1g-dev libboost-all-dev libasound2-dev pulseaudio libopencv-dev libxml2-dev bison flex \
ffmpeg libavcodec-dev libavformat-dev libopus-dev doxygen graphviz

srcejon, 2022-10-01:
Fixed, thanks

closed 2022-10-01 by srcejon