f4exb/sdrangel #1461: Fix debian package dependencies regarding PulseAudio

RAD750, 2022-10-02:
Hello,
Lately, several Ubuntu/Debian based distros, such as Pop!_OS among others, have started using PipeWire, which is a dropin replacement for PulseAudio.
However, the package explicitly requires the `pulseaudio` package to be installed, which will seriously break any distro that uses PipeWire as both packages cannot be installed simultaneously. (Usually dpkg just complains about unmet dependencies, but in one case after using the -f switch, it attempted to remove a whole host of packages including `pop-desktop` which would seriously damage the desktop environment)

It is possible to install SDRAngel on such a distro by unpacking the Debian package, modifying the control file by removing `pulseaudio`, and installing it. Everything then works as expected, sound included.

I'd suggest to replace `pulseaudio` in the control file, by `pulseaudio | pipewire-pulse`, that way SDRAngel can be installed both on distros using PulseAudio as well as PipeWire. (`pipewire-pulse` is the PulseAudio component of PipeWire).

Thank you for your time and software :smile:

f4exb, 2022-10-02:
This pipewire thing does not seem to be very mature. How does it fit with Qt audio?

RAD750, 2022-10-03:
Well, Ubuntu will switch to PipeWire in the next version (22.10); Pop!_OS already did that as did Fedora. 
It is a drop in replacement for PulseAudio so nothing needs to be done on your side, except removing it from the dependencies.

If the package stays that way with pulseaudio as a dependency, it will not be possible to install sdrangel on Ubuntu 22.10 when it comes out.

f4exb, 2022-10-03:
I am not keen on jumping on the latest distribution just because it is the latest. There should be a good reason for such a move. However if this change makes it compatible with future releases without compromising the LTS why not. However be warned that you will be on your own if you have any other issue with the distributed .deb on any other distribution than Ubuntu 22.04

RAD750, 2022-10-03:
Well if you do not have explicite requirements to stay on the LTS (and most people don't) it's usually a good idea to keep up to date with new software, especially drivers. 
The change does not in any way compromise anything, you could as well remove PulseAudio as a dependency and it would still work, who would install a distro without an audio server? 
 
The package for Audacity [does not list](https://packages.debian.org/bullseye/audacity) any audio server as a dependency, so I don't even know why this is a problem
Also, in your instructions for compiling for source, you should remove pulseaudio in the `apt install` command, as this will break a system that uses PipeWire.

By the way, if you officially support Fedora, then you do also support by extension PipeWire, as it is used by Fedora 34 and up since 2021.

f4exb, 2022-10-03:
the one and only supported system is Ubuntu 22.04 period

hiperiondev, 2022-10-03:
> This pipewire thing does not seem to be very mature. How does it fit with Qt audio?

My experience with pipewire started more than a year ago. With the first change in Fedora it works VERY badly: unexpected crashes, bad bluetooth integration, etc. I uninstalled it and continued using pulse audio.
Since Fedora 36 it's stable and, so far, I haven't had any problems. I use KDE so I assume that the integration with QT is correct

f4exb, 2022-10-05:
>you should remove pulseaudio in the apt install command

Of course not! Pulse audio is an essential piece of the whole thing and instructions are given in this context. You are free to make your own recipe but don't complain if anything breaks.

RAD750, 2022-10-05:
No, you're very wrong. Your package setup is wrong.

There is no need to specify the `pulseaudio` package as a dependency, or any other audio server for that matter.. It is not a dependency. Your software does not depend on the `pulseaudio` package being installed either for compilation or for runtime, it depends on there being an audio server just to get the audio out. It is very well possible to run your software, and any other sane Linux software, without a local PulseAudio server or with an alternative implementation of a PulseAudio server like `pipewire-pulse`. 

Take a look at any other software that works using PulseAudio, like [Audacity](https://packages.debian.org/bullseye/audacity), [VLC](https://packages.debian.org/bullseye/vlc), [mplayer](https://packages.debian.org/bullseye/mplayer), [GQRX](https://packages.ubuntu.com/jammy/gqrx-sdr), etc. None of these packages have `pulseaudio` as a dependency in their package, and there's a very good reason NOT to do what you are doing. I've linked the dependency lists of those packages so you can check yourself that I'm not making stuff up. And this is especially since you only support Ubuntu 22.04, which already comes with PulseAudio by default, so what's the point of adding it as a dependency? Are you afraid that people will install it on a server installation and complain that audio isn't working?

Seriously, before telling to people that they're trying to help that they're wrong, take a look on how other Debian software is packaged.

> the one and only (Linux) supported system is Ubuntu 22.04 period
And by the way, what is the point of advertising a Fedora package in **your** official wiki, if you say that it is not supported? Delete it from your wiki then if you don't want to deal with it.

RAD750, 2022-10-05:
Oh, and a follow up:
The proof that pulseaudio is not needed in the dependencies is in the fact the software works perfectly even without it.
[image]

f4exb, 2022-10-05:
That's complete rubbish! Most people will want to get audio out of this software.

f4exb, 2022-10-05:
Now if you think that this software is not sane or I am a moron nothing forces you to use this software or keep posting here.

closed 2022-10-05 by f4exb