f4exb/sdrangel #1463: APT problem?

loewal, 2022-10-03:
During startup of SDRangel, this is displayed:
2022-10-03 09:30:20.390 (W) PluginManager::loadPluginsDir: Cannot load library /usr/local/lib/sdrangel/plugins/libdemodapt.dylib: (dlopen(/usr/local/lib/sdrangel/plugins/libdemodapt.dylib, 0x0085): Symbol not found: (_apt_calibrate_thermal)
  Referenced from: '/usr/local/lib/sdrangel/plugins/libdemodapt.dylib'
  Expected in: '/usr/local/lib/sdrangel/libapt.dylib')
I recompiled APT as described.
Any help?

srcejon, 2022-10-03:
Are you sure the new version was installed to /usr/local/lib/sdrangel/libapt.dylib?

What exact commands did you build with?

loewal, 2022-10-04:
Found a smaller one in /usr/local/lib/sdrangel and a bigger one in /usr/local/lib.
Copied last one in sdrangel and the warning was gone.
I followed these instructions:

git clone https://github.com/srcejon/aptdec.git
cd aptdec
git checkout libaptdec
git submodule update --init --recursive
mkdir build; cd build
cmake -Wno-dev ..
make 
sudo make install

That's not right, isn't it?

From these instructions (partly)
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

srcejon, 2022-10-04:
The bit you're missing

-DCMAKE_INSTALL_PREFIX=/opt/install/aptdec

Is the important part, as you're not controlling where it is installed. This needs to match up with the path you use when running sdrangel's cmake.

It doesn't need to be the above path, but you need to make sure it's built wherever you are telling sdrangel it is.

closed 2022-12-09 by github-actions[bot]