f4exb/sdrangel #1469: Snap Store Version: "Illegal instruction" Crash

0brym, 2022-10-08:
Installed SDRAngel from the Snap store (latest/stable v5.9.0) on Ubuntu 20.04.5.

Every time I run the sdrangel command from the Terminal, it immediately crashes out, saying: Illegal instruction (core dumped)

Also, system doesn't show any results when searching for sdrangel under Show Applications (press Super (windows start button on keyboard) > start typing), though it does show as installed through the Ubuntu Software application. So no shortcuts were created during installation either.

srcejon, 2022-10-08:
What CPU do you have? Is it quite old? It may be missing some SSE instructions.

Why not install one of the more recent releases from github? V5 is old.

0brym, 2022-10-08:
Had no idea it was old. Just saw on a website that it was available as a snap. Went with it.

Just checked my CPU flags, and both sse4_1 and 4_2 are listed. I'll get rid of the snap since it clearly doesn't work.

srcejon, 2022-10-08:
I think this is the most recent version for 20.04: https://github.com/f4exb/sdrangel/releases/download/v7.6.4/sdrangel-1670-master.tar.gz

f4exb, 2022-10-08:
> Had no idea it was old. Just saw on a website that it was available as a snap

Please do not trust unknown sources,,, Moreover Snap is a piece of s**t.

0brym, 2022-10-08:
With respect, the snap I installed had your github page linked. There was no reason at all to believe it was not a trusted source. So you should probably address that yourself. That's your responsibility given it's your name on the tin.

Secondly, this is the only s**t experience I've ever had with snaps. But to each their own.

f4exb, 2022-10-08:
Please read the Wiki more carefully: https://github.com/f4exb/sdrangel/wiki#snap-and-flatpak

This is Open Source and ultimately the only trusted source is the source code itself. I am not responsible of all the bulls**t people may make based on this source. Conversely some other people may do better with it and be more honest with their users. This is the deal. In any case I always do recommend to compile the source code yourself not only this is more secure because the code is visible to everyone  but it will also ensure that the result will best fit your system. And should you find any issue it would probably best fit as an issue here than reporting some obscure artifact failing for a reason I cannot care less about. Anyway "Illegal instruction" is a clear indication that the artifact does not fit in your system.

The few binary artifacts you will find here in the releases section are products of a CI/CD chain and are therefore in close control. They are here for convenience and this does not change my recommendation above.

Finally of course my opinion on Snap is my opinion obviously based on an experience different than yours. Believe me Snap, Flatpack, Appimage and the like fall short of the objective of making an application run as an independent bundle.It may work... or not. Only Docker is up to the task. The first thing I do with a fresh installation of Ubuntu is to remove Snap as a useless burden on my system.

0brym, 2022-10-09:
Thanks for your help, srcejon. Much appreciated.

closed 2022-12-14 by github-actions[bot]