f4exb/sdrangel #1470: BUG: In "Satellite Tracker" AOS/LOS commands, quoted strings are NOT passed as a single argument

gvimlag, 2022-10-08:
PROBLEM:
In the "Satellite Tracker" AOS/LOS commands, strings enclosed in quotes ("this is a test") are NOT passed as a single argument.
Instead, the quoted strings are passed as multiple arguments.
```
Example: "this is a test" is passed as: $1=this           $2=is      $3=a       $4=test
         instead of:                    $1=this is a test $2=<empty> $3=<empty> $4=<empty>

```
TEST SCRIPT ("/tmp/test.bash"):
```
#!/bin/bash
echo \$1=$1   > /tmp/test.txt
echo \$2=$2  >> /tmp/test.txt
echo \$3=$3  >> /tmp/test.txt
echo \$4=$4  >> /tmp/test.txt
```

Run from command line: /tmp/test.bash "this is a test"
```
cat /tmp/test.txt
$1=this is a test
$2=
$3=
$4=
```

Run from "Satellite Tracker"->"SDRangel control"->"AOS command": /tmp/test.bash "this is a test"
```
cat /tmp/test.txt
$1=this
$2=is
$3=a
$4=test
```

REQUEST:
In the Satellite Tracker AOS and LOS commands (under both "Show settings dialog" and "SDRangel control"), please pass quoted strings as a single argument.

srcejon, 2022-10-08:
Which version are you using, as I thought I fixed this for 7.8.0?

gvimlag, 2022-10-08:
I'm using version 7.7.0-58.

I see that a new version was posted 2 days ago.  (I didn't notice the update before posting.  Sorry.)
I'll install it and see what happens.

gvimlag, 2022-10-09:
Confirmed: Version 7.8.0-10-gc6496b11f does NOT have this quoted string bug.
I withdraw my bug report.
Thank you for your work.

Unfortunately, I've discovered a new issue with the AOS/LOS command.
The "AOS command" does not accept multiple commands.
For example: /tmp/test.bash "this is a test" ; /tmp/test.bash this is a test

TEST SCRIPT ("/tmp/test.bash"):
```
#!/bin/bash
echo \$1=$1  >> /tmp/test.txt    #changed from overwrite (">") to append (">>")
echo \$2=$2  >> /tmp/test.txt
echo \$3=$3  >> /tmp/test.txt
echo \$4=$4  >> /tmp/test.txt
```

Run from command line: /tmp/test.bash "this is a test" ; /tmp/test.bash this is a test
```
cat /tmp/test.txt
$1=this is a test
$2=
$3=
$4=
$1=this
$2=is
$3=a
$4=test
```

Run from "Satellite Tracker"->"SDRangel control"->"AOS command": /tmp/test.bash "this is a test" ; /tmp/test.bash this is a test
```
cat /tmp/test.txt
$1=this is a test
$2=;
$3=/tmp/test.bash
$4=this
```

REQUEST:
Please allow the use of ";", "&&" and "||" to chain commands in the AOS/LOS command.
Example-1: /tmp/test.bash "this is a test" ;  /tmp/test.bash this is a test
Example-2: /tmp/test.bash "this is a test" && /tmp/test.bash this is a test
Example-3: /tmp/test.bash "this is a test" || /tmp/test.bash this is a test

srcejon, 2022-10-09:
Sorry, but I won't try to add support for that. If you want to run multiple scripts, you should be able to write a script to do it.

closed 2022-12-14 by github-actions[bot]