f4exb/sdrangel #1475: Feature Request: HD Radio Demodulator

CRD716, 2022-10-12:
https://www.sigidwiki.com/wiki/HD_Radio_(FM)
https://en.wikipedia.org/wiki/HD_Radio
More importantly (and usefully)
https://github.com/theori-io/nrsc5

This is a very popular standard in my area, and it would be nice to have the option of high quality FM (and AM) stations with an SDR dongle. From my basic investigation it seems like it is mostly proprietary, however the github linked and sididwiki seem to point to there being at least some information out there.

Side note: Sorry for opening all these feature requests, just putting some ideas out there.

srcejon, 2022-10-13:
Ideas welcome - but only seems to be in North America, so not something I can look at. Maybe someone else though.

CRD716, 2022-10-13:
I'll attempt to grab some iq samples when I have the time, hopefully they'll help.

CRD716, 2022-10-28:
[I think this should be good enough, was messing with some things though.](https://drive.google.com/file/d/16STXA8evVZt50Pg_yu6fWm-78y1Cq3BK/view?usp=sharing)

closed 2023-01-02 by github-actions[bot]