f4exb/sdrangel #1477: Remove non significant upper digits on spectrum frequency marker ticks

f4exb, 2022-10-15:
Follow up on #1476 

Here is the issue:
[image]

When the current frequencies are way above the bandwidth here 15.625 kHz at ~1 GHz the most significant digits are actually... not significant and make it values hard to read unless you count the digits. In this example the 1.00000's are encumbering the display.

This issue is opened to optionally remove these leading digits when possible. A possible algorithm is proposed as follows:

Example knowing the start and end frequency of the scale and thus its width
Start: 10,496,750,000 Hz
End: 10,499,750,000 Hz
=> Width 3,000,000 Hz

So I should display at least up to the 10⁶ digit. Now I will move through the upper digits and "drop" the lower digits until start and end matches. Here it is very quick since at the next on (10⁷ ) we have 10,490,000,000. So just subtract this value and you obtain a displayable range of 6,750,000 to 9,750,000.

If there was any boundary crossed at an upper digit the start and end would not match and you will have to take at least up to that digit. Other example
Start 10,499,500,000 Hz
End: 10,500,500,000 Hz
=> Width: 1,000,000 Hz

So the algorithm starts at 10⁶ and the first matching start/end is at 10⁹ so you know you have to subtract 10,000,000,000 and thus end up with a display scale between 499,500,000 and 500,500,000 to which the usual formatting algorithm would apply. You could add a leading tick (') to mark that the upper digits have been dropped (similar to what is done in the BATC wideband monitor).

Of course this works only if the start and end have the same int(log10(x)). But if this is not the case then this algorithm simply does not apply and numbers have to be considered in full.

srcejon, 2022-10-15:
A couple of additional random thoughts:

- To make larger numbers readable, I think grouping digits into 3 really helps. E.g 1.495,780 instead of 1.49578. Takes a bit more space, but a good trade-off for those with plenty of space - could be a user-specified option.
- For the abbreviated form you describe, I'm wondering if the units might be too non-obvious, after the usual formatting algorithm is applied. E.g. Would 6,750,000 be displayed as just 6.75? If so, perhaps the units k/M should be displayed.
- Perhaps the omitted digits could be displayed on the left hand-side of the scale or just for the first tick, if there is space? Might help - might not.

f4exb, 2022-10-15:
It has to be systematic and simple. This touches part of the code that are critical and the most old and thus quite brittle.

f4exb, 2022-10-16:
It is still easily possible to display the multiplier string e.g.'6.75M instead of just '6.75

f4exb, 2022-10-16:
This is what you would obtain in the conditions of the screenshot on the top:
[image]
Which is quite fine for me. Notice the increase in grid resolution.

commit da8e4d1a (2022-10-16): Spectrum frequency ticks MSB truncation: fixed algorithm, settings persistency and added documentation. Fixes #1477
closed 2022-10-16 by f4exb