f4exb/sdrangel #1480: Ubuntu 22.04 w/ 7.8.0 missing M17

alphafox02, 2022-10-16:
Finally getting around to testing SDRAngel more in 22.04, using the Linux Install from Source page I made sure all the Apt related packages are installed. Grabbed the latest Debian Deb, installed, but after loading there's no M17 RX/TX options. 

Is this expected? With 20.04 and the last release before the change in QT, both options were available. I've been looking at the output in terminal, but so far only noticed this.

`Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libmbe.so.1: cannot open shared object file: No such file or directory)
`
I've looked at the output a couple times and see not even a mention of the m17 library attempting to be loaded.

alphafox02, 2022-10-16:
Same issue applies to 7.7.0 Deb (SDRAngel-1696-master).

srcejon, 2022-10-16:
Have you got codec2 installed? Is it found?

Might want to post cmake output / CMakeCache.txt

alphafox02, 2022-10-16:
This was using the Deb package on the releases page, not built from source. 

Next step would be to remove deb package and build from source. I only mentioned the Linux page since that's where I pulled the apt packages from prior to installing the deb. Sorry if that wasn't clear on that part.

alphafox02, 2022-10-16:
You were correct on codec2 not being installed (via the package manager). However, that did not help with the deb package and no rx/tx M17 packages being listed.

alphafox02, 2022-10-16:
What's showing in /usr/lib/sdrangel/plugins after deb install on 22.04, which has libmodemm17.so shown. 

[image]

alphafox02, 2022-10-16:
I checked in a 20.04 built from source and the plugins folder contains libmodm17.so, so it appears the 22.04 deb package provided on the releases page is missing some things.

alphafox02, 2022-10-16:
Sat down, built SDRAngel from source on 22.04 and neither issue I mentioned above is there. There's an issue with the deb releases best I can tell. 

In the source build the lib file is in the plugins folder for M17 and both rx/tx are present in the GUI. This is not the case for the deb package. There's also no complaint about libmbe.so.1 in the source built SDRAngel.

sp5wwp, 2022-10-17:
Thanks for investigating this.

f4exb, 2022-10-18:
For DAB this line appears in the CMake output:
```
[00:20:19] -- Can't build DAB library as missing zlib or faad
```

f4exb, 2022-10-18:
Well maybe this is here: https://github.com/f4exb/sdrangel/blob/master/.appveyor.yml#L94 I thought it would go through external/CMakeLists.txt but apparently it does not. I'll have to revert a few things...

f4exb, 2022-10-18:
In fact the issue may be here:
https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L204
and here...
https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L286
https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L308
https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L591

Because we force the use of packaged libraries that may or may not be installed and may or may not be reported properly.

I guess in the case of codec2 that `CODEC2_FOUND` is false and since `USE_PRECOMPILED_LIBRARIES` is true the external project is not built.

f4exb, 2022-10-18:
Apparently it needs `ENABLE_EXTERNAL_LIBRARIES=AUTO`

The latest commit builds M17 and FreeDV plugins. Still not DAB: https://ci.appveyor.com/project/f4exb/sdrangel/build/job/i8n22vgrsk8mc27x

f4exb, 2022-10-18:
In the log of CMake I see:
```
[00:19:06] -- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
...
[00:19:06] -- Found FAAD: /usr/lib/x86_64-linux-gnu/libfaad.so  
```
So there should be no reason not to build libdab...

alphafox02, 2022-10-19:
I noticed in appveyor build finished about 2 hours ago that M17 is there, but looking at output in terminal I think maybe a bunch of other things are missing somehow now. Just pasting what I see, maybe still a work in progress. Strange, wish I could be of better help but I'm not familiar with the external libraries settings and such. 

```
 Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libdsdcc.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libfeatureambe.so: (libserialdv.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libremotesink.so: (libcm256cc.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libremotesource.so: (libcm256cc.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/liboutputremote.so: (libcm256cc.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/liboutputlimesdr.so: (libLimeSuite.so.20.01-1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libmimolimesdr.so: (libLimeSuite.so.20.01-1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libinputremote.so: (libcm256cc.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libinputperseus.so: (libperseus-sdr.so.0: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libinputlimesdr.so: (libLimeSuite.so.20.01-1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libinputairspyhf.so: (libairspyhf.so.0: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libfeaturelimerfe.so: (libLimeSuite.so.20.01-1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libfeatureambe.so: (libserialdv.so.1: cannot open shared object file: No such file or directory)
Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libdsdcc.so.1: cannot open shared object file: No such file or directory)
```

f4exb, 2022-10-19:
I think we should revert to EXTERNAL_LIBRARIES=ON and make it a "no" here: https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L204

alphafox02, 2022-10-19:
I think you got it, just downloaded 1805 and took a quick peek. M17 is there and I see no complaints of cannot load in the terminal output.

N5BRG, 2022-10-19:
I have been working to build 7.8.0 on Ubuntu 22.04 also.

My computer is Dell 3060   I5 with 16G of memory
***@***.***:~/sdrangel/build$ lsb_release -a
LSB Version: core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID: Ubuntu
Description: Ubuntu 22.04.1 LTS
Release: 22.04
Codename: jammy

I see in the docs for strangle we need C++14

I see in CMakeList.txt:
set(CMAKE_CXX_STANDARD 14)

On my system I see
c++  —version:
c++ (Ubuntu 10.3.0-15ubuntu1) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I started with c++ version 11 until I started trying to get it up to c++14.

For reference I get this:
***@***.***:~$ sudo update-alternatives --config g++
[sudo] password for bob:
There are 2 choices for the alternative g++ (providing /usr/bin/g++).

  Selection    Path             Priority   Status
------------------------------------------------------------
* 0            /usr/bin/g++-10   10        auto mode
  1            /usr/bin/g++-10   10        manual mode
  2            /usr/bin/g++-9    9         manual mode

Press <enter> to keep the current choice[*], or type selection number: 0
***@***.***:~$ sudo update-alternatives --config gcc
There are 2 choices for the alternative gcc (providing /usr/bin/gcc).

  Selection    Path             Priority   Status
------------------------------------------------------------
* 0            /usr/bin/gcc-10   10        auto mode
  1            /usr/bin/gcc-10   10        manual mode
  2            /usr/bin/gcc-9    9         manual mode

Press <enter> to keep the current choice[*], or type selection number: 0

According to what I have read C++14 is supported in gcc-9 and up.

Do others feel the proper c++ is being used when building sdrangle.
Can someone that gets a clean build post what they have for: sudo update-alternatives --config gcc   ? Hopefully on an x86 system.

Bob N5BRG

f4exb, 2022-10-19:
Actually I have reverted to original https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt#L204 but changed appveyor.yaml so that it does not try to use the shell scripts to build some dependencies. I also use `DENABLE_EXTERNAL_LIBRARIES=AUTO` so that the proper find packages are invoked.

f4exb, 2022-10-19:
@N5BRG this is off topic please post to https://groups.io/g/sdrangel

f4exb, 2022-10-19:
Implemented in v7.8.1

closed 2022-10-19 by f4exb